openwrt/packages #24880: is this normal? obfs4proxy RAM load 500%

nitro3188, 2024-08-24:
For some reason, the Xiaomi AX3200 router freezes after 5 minutes after the start (when the databases start to be updated)... Ruantiblock, HTTPS DNS Proxy, Adblock were installed - previously all this worked without a hint of bugs, now I had to disable Adblock - although the databases are sparing - it does not hang in this form yet. 211mb of 240mb of RAM is used (with Adblock disabled). There is generally something scary about obfs4proxy processes "2557 tor /usr/bin/obfs4proxy 0% 501%"
I attached a screenshot...
[image]

brada4, 2024-08-24:
Please add
```
ubus call system board
cat /proc/2557/status
```

nitro3188, 2024-08-24:
[image]
here is, please

brada4, 2024-08-24:
Please post TEXT, should have come obvious that I  would like status of 500% process PID, number in example is from your first picture assuming you dont reboot every minute.

nitro3188, 2024-08-24:
9663 | tor | /usr/bin/obfs4proxy | 0% | 501%

root@OpenWrt:~# ubus call system board
{
        "kernel": "5.15.162",
        "hostname": "OpenWrt",
        "system": "ARMv8 Processor rev 4",
        "model": "Xiaomi Redmi Router AX6S",
        "board_name": "xiaomi,redmi-router-ax6s",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.4",
                "revision": "r24012-d8dd03c46f",
                "target": "mediatek/mt7622",
                "description": "OpenWrt 23.05.4 r24012-d8dd03c46f"
        }
}
root@OpenWrt:~# cat /proc/9663/status
Name:   obfs4proxy
Umask:  0022
State:  S (sleeping)
Tgid:   9663
Ngid:   0
Pid:    9663
PPid:   1939
TracerPid:      0
Uid:    52      52      52      52
Gid:    52      52      52      52
FDSize: 64
Groups: 52 
NStgid: 9663
NSpid:  9663
NSpgid: 1
NSsid:  1
VmPeak:  1233164 kB
VmSize:  1233164 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:     11928 kB
VmRSS:      5808 kB
RssAnon:            2808 kB
RssFile:            3000 kB
RssShmem:              0 kB
VmData:    46724 kB
VmStk:       132 kB
VmExe:      4844 kB
VmLib:       720 kB
VmPTE:        88 kB
VmSwap:        0 kB
CoreDumping:    0
THP_enabled:    0
Threads:        10
SigQ:   0/956
SigPnd: 0000000000000000
ShdPnd: 0000000000000000
SigBlk: fffffffc3bba3a00
SigIgn: 0000000000000000
SigCgt: fffffffc7fc1feff
CapInh: 0000000000000000
CapPrm: 0000000000000000
CapEff: 0000000000000000
CapBnd: 000001ffffffffff
CapAmb: 0000000000000000
NoNewPrivs:     0
Seccomp:        0
Seccomp_filters:        0
Speculation_Store_Bypass:       not vulnerable
SpeculationIndirectBranch:      unknown
Cpus_allowed:   3
Cpus_allowed_list:      0-1
Mems_allowed:   1
Mems_allowed_list:      0
voluntary_ctxt_switches:        13
nonvoluntary_ctxt_switches:     5
root@OpenWrt:~# opkg list-installed | grep obfs

brada4, 2024-08-25:
Can you read it ?

nitro3188, 2024-08-25:
you said that you need text, not a photo... if you mean crossed out text, then that's how the forum handles symbols, I copied it as is from the terminal

brada4, 2024-08-25:
Bug tracker is not a support forum. 
Notice formatting buttons above edit box?
Your friend is called `</>`

nitro3188, 2024-08-25:
got it!

brada4, 2024-08-25:
vmsize is address space used, actual on-chip ram consumption is the sum of the small numbers that follow.  Either you write another report in ../luci that it visualises little relevant parameter or wait months for somebody to notice and move this report over there and act upon.

Actual usage is sum(others) ~75MB just that process allocated but never touched much bigger memory chunks. Since address space on 32bit system is <4GB you may run out of that per process vs running out of actual RAM.

nitro3188, 2024-08-25:
> Either you write another report in ../luci that it visualises little relevant parameter

If you tell me more about how to do this, I will be very grateful.

brada4, 2024-08-25:
Like here (option 3 luci)
https://openwrt.org/bugs