openwrt/packages #24881: "games_tracking" list blocks unrelated websites

DenebTM, 2024-08-25:
`games_tracking` completely blocks various websites such as `bild.de`, `www.engadget.com`, `www.facebook.com`, `instagram.com`, `taz.de`, as per the list at https://raw.githubusercontent.com/KodoPengin/GameIndustry-hosts-Template/master/Main-Template/hosts.

**Note:** even the [gaming-specific list](https://raw.githubusercontent.com/KodoPengin/GameIndustry-hosts-Template/master/Gaming-Full-Template/hosts) by the same author contains multiple entries for `www.facebook.com` and `www.instagram.com`. Might be better off finding a different source for this blocklist altogether.

brada4, 2024-08-28:
It is your decision to use list or not. As the url says list is not compiled by OpenWRT. Ask authors after liberating your favourite game from metverse widgets.

dibdot, 2024-08-29:
Please report it upstream to the list maintainer.

aphorise, 2024-09-20:
These comments were very helpful & I was having similar issues particularly on livestreams from YouTube so I resorted to disabling this seed for now.
@DenebTM did you raise a request upstream & if so can you reference / link it here as well as there for the wider community?

DenebTM, 2024-10-07:
I have now raised an issue with the author about the somewhat misleading README, but imho this is not _really_ their problem, @dibdot — they do mention that e.g. "tech companies like Facebook are blocked", which I would simply not expect from a blocklist entry named "games_tracking".

(Granted, back in 2020 when you added the list, the README *did not* mention it.)

brada4, 2024-10-07:
Facebook does games and tracking. Do you think double denial means permit?

KodoPengin, 2025-01-01:
Since it's about my lists, a brief response here, even if the topic has already been addressed.

As mentioned, Facebook (META) track in games, regulary software and IOT. Beginning from the main domain. In the mobile sector, even more so than with desktop applications.

In addition, there are numerous data protection violations, financial penalties and data leaks each year. ALso published an article about Facebook on my project page and there are also numerous examples of tested products including raw data.

https://github.com/KodoPengin/GameIndustry-hosts-Template/issues/43

May give false positive warnings but everything's clean
https://gameindustry.eu/blog/why-facebook-is-blocked/
https://gameindustry.eu/katalog/meta/

As mentioned, you (and anyone else) are free to use my lists.