openwrt/packages #24908: package: redundant downloading package tarballs

ikazuhiro, 2024-09-01:
### Describe the bug

Sine https://github.com/openwrt/openwrt/commit/bab3ae2ee7656600a185f4cef11cef94389023af, OpenWrt prefers .gz tarball than .xz for tools.  But packages in `package` directory and [package repository](https://github.com/openwrt/packages) still seem to prefer .xz.  As a result, `dl` directory contains many tarballs which have the same content.  For example, I can see following redundant tarballs in my `dl` directory.
```
autoconf-2.71.tar.gz
autoconf-2.71.tar.xz
automake-1.16.5.tar.gz
automake-1.16.5.tar.xz
e2fsprogs-1.47.0.tar.gz
e2fsprogs-1.47.0.tar.xz
expat-2.6.2.tar.gz
expat-2.6.2.tar.xz
libtool-2.4.7.tar.gz
libtool-2.4.7.tar.xz
patch-2.7.6.tar.gz
patch-2.7.6.tar.xz
zlib-1.3.1.tar.gz
zlib-1.3.1.tar.xz
```

### OpenWrt version

master

### OpenWrt release

master

### OpenWrt target/subtarget

all

### Device

host

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

BKPepe, 2024-09-01:
> As you can see in the linked [openwrt/openwrt@bab3ae2](https://github.com/openwrt/openwrt/commit/bab3ae2ee7656600a185f4cef11cef94389023af) the .gz or .bz2 archives are now preferred, thus if something still downloads `autoconf-2.71.tar.xz` then its a bug and if you want that fixed, you need to provide more details, like for example reproducer or complete log.

Is this your opinion, or is this the official statement from OpenWrt core members? I've asked on the mailing list [^1] about this, but no one responded to it except @robimarko. For me, it makes sense to get away from .tar.xz, but other package maintainers are not united about this and as you can see, recently in PR https://github.com/openwrt/packages/pull/24299/files, it was changed from ``.tar.gz`` to ``tar.gz`` for several packages without proper reason, though except that ``.tar.xz`` is preferred and mentioned in one file. 🤷 . 

[^1]: https://lists.openwrt.org/pipermail/openwrt-devel/2024-June/042933.html

diizzyy, 2024-09-24:
I don't really see the issue with tar.xz tarballs especially since there have been lots of eyes on xz-utils since. Just grab whatever is the smallest tarball and go with it, it's not something other distros have enforced either to my knowledge.

BKPepe, 2024-10-02:
No conclusion was made in the referenced PR (iputils), where it was also discussed. Closing.

closed 2024-10-02 by BKPepe