openwrt/packages #24999: pbr: start fails if dynamic routing is enabled (frr ospfd)

softengel, 2024-09-19:
Maintainer: @stangri
Environment: OpenWrt 23.05.4 r24012-d8dd03c46f arch x86_64, pbr Version 1.1.6-20 - Running (fw4 nft file mode),

Description:
After setup of frr and opsfd pbr service was not able to start.
pbr iterates over main routing table and tries to copy routes from main table to pbr_$interface route tables but fails with dynamic ospf routes.

My fix for this issue is to copy only static routes. Now pbr is working for my WireGuard interfaces. 

Better solution is welcome. I don't know if somebody else get bad side effects with this fix in more complex setups of pbr.
Would be nice to have a configuration value for pbr to filter for static routes only if my fix is not a global solution.

```
diff /etc/init.d/pbr /etc/init.d/pbr.org
1740c1740
<                                       $(ip -4 route list table main proto static)
---
>                                       $(ip -4 route list table main)
```

stangri, 2024-09-22:
@softengel please help me understand the issue a bit better, wouldn't `ip -4 route list table main` also include the output produced by `ip -4 route list table main proto static`?

stangri, 2024-09-23:
@softengel thanks for your feedback, I'll introduce your fix in 1.1.7-11 and see if it produces unwanted side effects.

softengel, 2024-09-24:
@stangri it sounds good :-)
Maybe consider to be not as strict as I and think about filtering only dynamic protocols ' ip route list table main | egrep -v " ospf | bgp | eigrp | rip " ' but it can be more like isis etc. If you know what is realy needed then something like ' ip route list table main proto kernel proto static ' would be better if you need local interface routes, too. Hope you will find a good solution.

closed 2024-09-24 by stangri