openwrt/packages #25014: iperf3 build failure

graysky2, 2024-09-23:
Maintainer: @nbd168 
Environment: x86/64, r27494

Description:  currently, iperf3 experiencing a build failure due to a missing dependency on `libsctp.so.1`

Partial build log:
```
...
Package libiperf3 is missing dependencies for the following libraries:
libsctp.so.1
make[2]: *** [Makefile:121: /scratch/union/bin/packages/x86_64/packages/libiperf3_3.17.1-r2_x86_64.ipk] Error 1
make[2]: Leaving directory '/scratch/union/feeds/packages/net/iperf3'
time: package/feeds/packages/iperf3/nossl/compile#10.14#1.54#12.18
    ERROR: package/feeds/packages/iperf3 failed to build (build variant: nossl).
make[1]: *** [package/Makefile:177: package/feeds/packages/iperf3/compile] Error 1
make[1]: Leaving directory '/scratch/union'
make: *** [/scratch/union/include/toplevel.mk:248: package/iperf3/compile] Error 2
```

Link to complete [build log](https://gist.github.com/graysky2/0882809eb925e34eb2f2d5963eae9aef).

krant, 2024-09-23:
@graysky2 thanks for the reporting, the fix has been pushed.

graysky2, 2024-09-23:
Thanks @krant, that was fast!

closed 2024-09-24 by BKPepe