openwrt/packages #25033: ddns-scripts: Fix updating dynu addresses

curuvar, 2024-09-25:
Apparently dynu.com has updated their API.

Instead of specifying an update with a fragment ...&hostname=my-host.my-domain.com&... the new API uses the fragment ...&hostname=my-domain.com&alias=my-host&...

curuvar, 2024-09-25:
I modified my  /usr/share/ddns/default/dynu.com.json to change:

   ...?hostname=[DOMAIN]&...

to

   ...?hostname=[DOMAIN]&alias=[PARAMOPT]&...

Then put the alias name in the PARAMOPT field when editing with luci.

Neustradamus, 2024-12-09:
cc: @giannoug.

alonsodomin, 2025-12-29:
Dynu's docs on the matter are somewhat fragmented and mostly don't mention the _alias_ update, except this one that is specific to using `wget`: https://www.dynu.com/en-US/DynamicDNS/IPUpdateClient/Wget

The thing is, at the moment if you have a custom domain in Dynu AND only want to update a subdomain (aka _alias_) keeping the root domain unchanged, the configuration options provided by OpenWRT will fail when trying to perform the IP update (logs show a `badauth` response even though OpenWRT thinks it actually succeeded):

```
222118       : #> /usr/bin/wget --hsts-file=/tmp/.wget-hsts -nv -t 1 -O /var/run/ddns/myddns_ipv4.dat -o /var/run/ddns/myddns_ipv4.err --ca-directory=/etc/ssl/certs --no-proxy 'https://api.dynu.com/nic/update?hostname=fqdn&myip=x.x.139.72&username=xxxxx&password=***PW***'
 222119       : DDNS Provider answered:
badauth
 222119  info : Update successful - IP 'x.x.139.72' send
 222119  info : Forced update successful - IP: 'x.x.139.72' send
```

The correct call should be instead like the following:
```
/usr/bin/wget "https://api.dynu.com/nic/update?hostname=custom_domain.tld&alias=subdomain&myip=x.x.139.72&myipv6=no&username=xxxxx&password=***PW***"
```
This will update the IP for `subdomain.custom_domain.tld` correctly while keeping the `custom_domain.tld` record unchanged.

I can confirm that this issue is still present today and the change suggested by @Uthedris fixes the issue by allowing to input the subdomain part in the `PARAMOPT` field in LuCI