openwrt/packages #25035: lpac: lpac-libmbim-wrapper doesn't work

bam80, 2024-09-25:
Maintainer: @blocktrron

Description:
`lpac.sh` wrapper is currently installed by the name of the binary in `/usr/bin/lpac`:
https://github.com/openwrt/packages/blob/b1635b899da0481867b0d04df1b007285721357f/utils/lpac/Makefile#L57

This creates problem as the wrapper overrides env variables which can be set by other upper wrappers.

_For now, https://github.com/stich86/lpac-libmbim-wrapper/ doesn't work._

`lpac.sh` must not mask real binary - instead, it should be installed separately.

blocktrron, 2024-09-26:
Wrapping the command is to enable usage with uci.

Looking at the project you are referring to, please make https://github.com/stich86/lpac-libmbim-wrapper/blob/main/lpac-mbim#L119C5-L119C8 compatible to supply a `--lpac-path` argument.

bam80, 2024-09-26:
> Wrapping the command is to enable usage with uci.

Still, the path is not required to be the same as the binary, right? Use whatever path for your wrapper, but do not touch the binary path, it's not yours.

I think the upstream project behavior is correct, and shouldn't need changes.
Any wrapper should be able to rely on standard path for the binary, adopting it just for the purpose of your hack shouldn't be needed.

If I'm missing something, please let me know.

bam80, 2024-09-26:
> Where should i put it if not in PATH?

the PATH is OK, just name it differently, I mean. Nobody will complain.

> you can check if the variables are pre-set in the environment and skip the assignment from UCI in the wrapper.

I do not see why would we want to undertake such complexity if the fix is as simple as rename your wrapper.
Still need to see what others say. @BKPepe 

Anyway, if you want to go that route, would you like to make the fix yourself?
Just make sure it's reviewed positively before merge, please.
Thank you.

robimarko, 2024-09-27:
I quite like the wrapper as otherwise you gotta do it all manually which is quite annoying

bam80, 2024-09-27:
> If what you require is to pass the command-line arguments to the wrapper and override the envrionment it modifies, you can check if the variables are pre-set in the environment and skip the assignment from UCI in the wrapper.

Please see #25046

robimarko, 2025-05-21:
There is no reason for the wrapper anymore as mbim support is now part of LPAC

closed 2025-05-21 by BKPepe

bam80, 2025-05-21:
@robimarko have you removed your wrapper from the packages then?

blocktrron, 2025-05-22:
No, because it supports  MBIM natively since https://github.com/openwrt/packages/commit/864d80aaeb14c304e437762fc8465db006fb02c2 as @robimarko already pointed out.

bam80, 2025-05-22:
While the particular lpac-libmbim-wrapper might be not actual already, the common issues mentioned were not addressed, so it might be a problem in the future, still.