openwrt/packages #25037: chrony: Config files are not updated to match UCI config

nroach44, 2024-09-26:
# Issue template (remove lines from top till here)

Maintainer: @BKPepe  (find it by checking history of the package Makefile)
Environment: ath79, Meraki MR18, 23.05.05

Description:
I've built my own image for the device, but no code changes, only to the OWRT config.

UCI doesn't seem to update the chrony config correctly, if at all:

```
root@OWRT-MR18:~# cat /etc/chrony/chrony.conf 
# Load UCI configuration
confdir /var/etc/chrony.d

# Load NTP servers from DHCP if enabled in UCI
sourcedir /var/run/chrony-dhcp

# Log clock errors above 0.5 seconds
logchange 0.5

# Don't log client accesses
noclientlog

# Mark the system clock as synchronized
rtcsync

# Record the clock's drift
driftfile /var/run/chrony/drift

# Save NTS keys and cookies
ntsdumpdir /var/run/chrony
root@OWRT-MR18:~# cat /var/etc/chrony.d/10-uci.conf 
pool 2.openwrt.pool.ntp.org maxpoll 12 iburst
makestep 1.0 3
nocerttimecheck 1
root@OWRT-MR18:~# cat /var/run/chrony-dhcp/lan.sources 
server <IP2> iburst
root@OWRT-MR18:~# uci export system.ntp
config timeserver 'ntp'
	list server '<IP1>'
	list server '<IP2>'
	list server '<IP3>'
	option use_dhcp '0'
```

As you can see here, the list of NTP servers is not reflected in the chrony config, as it only has "<IP2>", which is the IP of the server that DHCP tells clients to use.

It looks like neither the "use_dhcp" option, nor the server list is being written into the config files.

mlichvar, 2024-09-26:
chrony has its own UCI configuration in /etc/config/chrony. The one in /etc/config/system is used for the minimal ntpd from busybox.

nroach44, 2024-09-26:
> chrony has its own UCI configuration in /etc/config/chrony. The one in /etc/config/system is used for the minimal ntpd from busybox.

I am reading the chrony config (`/etc/chrony/chrony.conf`) in the above. Just to check, I checked `uci export`, but that only lists the package, no specific config:

````
uci export | grep -i chrony
package chrony
````

mlichvar, 2024-09-26:
Try `uci show chrony`.

jauderho, 2024-11-04:
Is there a way to get uci to show all the options it will recognize? For example, I tried setting xleave in /etc/config/chrony and it does not get written out to /var/etc/chrony.d/10-uci.conf

mlichvar, 2024-11-04:
The following command should print all supported options. You would need to look at the script to see in what section is each option accepted:
````
grep config_get /etc/init.d/chronyd
````

I think the default config shows all currently supported options. That might change with #23791 .

jauderho, 2024-11-05:
This is what I get when I run `grep config_get /etc/init.d/chronyd` against OpenWRT 23.05.5

```
	config_get_bool disabled "$cfg" disabled 0
	[ -z "$hostname" ] && config_get hostname "$cfg" hostname
	config_get minpoll "$cfg" minpoll
	config_get maxpoll "$cfg" maxpoll
	config_get_bool iburst "$cfg" iburst 0
	config_get_bool nts "$cfg" nts 0
	config_get iface "$cfg" interface
	config_get threshold "$cfg" threshold
	config_get limit "$cfg" limit
	config_get_bool rtccheck "$cfg" rtccheck 0
	config_get_bool systemcerts "$cfg" systemcerts 1
	config_get trustedcerts "$cfg" trustedcerts
```

Looking at the PR you referenced, it'll add `maxsources` if merged but not `xleave` and others. I'll see if I can find some time to add a PR at some point.