openwrt/packages #25052: ERROR: package/feeds/packages/ruby [host] failed to build.  ruby 3.3.4

smallprogram, 2024-09-29:
### Describe the bug

Operating System: Ubuntu 22.04.5
Memory: 16GB
Space: 90GB

compiling through github action, will report this error, ruby ​​cannot be compiled. Same process.

The error log in github action is as follows:

```log
Applying /workdir/openwrt/feeds/helloworld/lua-neturl/patches/010-userinfo-regex.patch using plaintext: 
x86_64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I..  -I. -I. -I../lib -I.. -I../libebl -I../libasm -I../libelf -I../libdw -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify    -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Wuse-after-free=3 -Werror -Wunused -Wextra -Wstack-usage=262144    -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/elfutils-0.191=elfutils-0.191 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -Wno-error=use-after-free  -c -o arm_retval.o arm_retval.c
patching file lib/net/url.lua
touch /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.prepared_5dbe9ef94015a441adb6e523f2881a91_6664517399ebbbc92a37c5bb081b5c53
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.configured_*
rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.lua-neturl_installed
(cd /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/./; if [ -x ./configure ]; then find /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1=neturl-1.1-1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="x86_64-openwrt-linux-musl-ld.bfd" NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1=neturl-1.1-1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1=neturl-1.1-1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.built_check
touch /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.built
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl.installed /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl
mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl/usr/lib/lua
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/lib/net/url.lua /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl/usr/lib/lua/
touch /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl.installed
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
x86_64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I..  -I. -I. -I../lib -I.. -I../libebl -I../libasm -I../libelf -I../libdw -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify    -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Wuse-after-free=3 -Werror -Wunused -Wextra -Wstack-usage=262144    -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/elfutils-0.191=elfutils-0.191 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -Wno-error=use-after-free  -c -o arm_cfi.o arm_cfi.c
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/.pkgdir/lua-neturl/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.lua-neturl_installed
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ipkg-all/lua-neturl
mkdir -p /workdir/openwrt/bin/targets/x86/64/packages /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ipkg-all/lua-neturl /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ipkg-all/lua-neturl/usr/lib/lua
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/lib/net/url.lua /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ipkg-all/lua-neturl/usr/lib/lua/
find /workdir/openwrt/build_dir/target-x86_64_musl/neturl-1.1-1/ipkg-all/lua-neturl -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
x86_64-openwrt-linux-musl-gcc -D_GNU_SOURCE -DHAVE_CONFIG_H -DLOCALEDIR='"/usr/share/locale"' -I. -I..  -I. -I. -I../lib -I.. -I../libebl -I../libasm -I../libelf -I../libdw -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify    -std=gnu99 -Wall -Wshadow -Wformat=2 -Wold-style-definition -Wstrict-prototypes -Wtrampolines -Wlogical-op -Wduplicated-cond -Wnull-dereference -Wimplicit-fallthrough=5 -Wuse-after-free=3 -Werror -Wunused -Wextra -Wstack-usage=262144    -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/elfutils-0.191=elfutils-0.191 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -D_GNU_SOURCE -Wno-unused-result -Wno-format-nonliteral -Wno-error=use-after-free  -c -o arm_initreg.o arm_initreg.c
make[4]: *** [Makefile:318: libruby-static.a] Aborted (core dumped)
make[4]: *** Deleting file 'libruby-static.a'
make[3]: *** [Makefile:1189: /workdir/openwrt/build_dir/hostpkg/ruby-3.3.4/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.3.4'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#167.75#25.40#201.85
    ERROR: package/feeds/packages/ruby [host] failed to build.
make[2]: *** [package/Makefile:175: package/feeds/packages/ruby/host/compile] Error 1
```

guhill1, 2024-10-14:
+1
我遇到相同问题,
https://github.com/immortalwrt/packages/issues/1336
目前是降级到3.25解决的,也不知道是大库的问题还是ruby的问题

luizluca, 2024-10-16:
> Can someone help me ?

I tried to reproduce this with openwrt in ubuntu 24.10 and I couldn't find a way to trigger it. Did you really build it using openwrt? I noticed some reports related to this issue at https://github.com/immortalwrt/packages/issues/1342 , which is not openwrt. Maybe something else there is triggering this error.

smallprogram, 2024-10-16:
> > 有人能帮我吗？
> 
> 我尝试在 ubuntu 24.10 中使用 openwrt 删除此问题，但找不到触发此问题的方法。您真的是使用 openwrt 构建的吗？我注意到[immortalwrt/packages#1342](https://github.com/immortalwrt/packages/issues/1342)中有与此问题相关的报告，但不是openwrt。也许还有其他原因触发了此错误。

I encountered this problem when I tried to build openwrt and immortalwrt in github action. Currently, I can compile by downgrading ruby, but the logs are out of date and cannot be found. Currently, there are only logs after downgrading.

closed 2024-11-06 by smallprogram

MrSorcus, 2025-07-13:
I can reproduce this on the main branch.
```
linking miniruby
/bin/sh ./tool/ifchange "--timestamp=.rbconfig.time" rbconfig.rb rbconfig.tmp
rbconfig.rb updated
generating encdb.h
./tool/generic_erb.rb:6:in 'Kernel#require': cannot load such file -- erb (LoadError)
	from ./tool/generic_erb.rb:6:in '<main>'
make[4]: *** [uncommon.mk:1261: encdb.h] Error 1
make[4]: Leaving directory 'openwrt/build_dir/hostpkg/ruby-3.3.6'
make[3]: *** [Makefile:1215: openwrt/build_dir/hostpkg/ruby-3.3.6/.built] Error 2
make[3]: Leaving directory 'openwrt/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#141.47#13.93#321.10
    ERROR: package/feeds/packages/ruby [host] failed to build.
make[2]: *** [package/Makefile:181: package/feeds/packages/ruby/host/compile] Error 1
make[2]: Leaving directory 'openwrt'
make[1]: *** [package/Makefile:177: openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory 'openwrt'
make: *** [openwrt/include/toplevel.mk:233: world] Error 2
```

Steps to reproduce
```
$ git clone https://github.com/openwrt/openwrt.git
$ cd openwrt
$ ./scripts/feeds update -a
$ ./scripts/feeds install -a
$ make menuconfig
$ ./scripts/diffconfig.sh
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq60xx=y
CONFIG_TARGET_qualcommax_ipq60xx_DEVICE_glinet_gl-axt1800=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libruby=y
CONFIG_PACKAGE_ruby=y
CONFIG_RUBY_ENABLE_YJIT=y
$ make -j 1 V=sc
```

smallprogram, 2025-07-14:
> I can reproduce this on the main branch.
>
> ```
> linking miniruby
> /bin/sh ./tool/ifchange "--timestamp=.rbconfig.time" rbconfig.rb rbconfig.tmp
> rbconfig.rb updated
> generating encdb.h [...]

```

MrSorcus, 2025-07-18:
> ```
> sed -i 's/--set=llvm\.download-ci-llvm=false/--set=llvm.download-ci-llvm=true/' feeds/packages/lang/rust/Makefile
> grep -q -- '--ci false \\' feeds/packages/lang/rust/Makefile || sed -i '/x\.py \\/a \        --ci false \\' feeds/packages/lang/rust/Makefile
> ```

And how is this supposed to help with `cannot load such file -- erb (LoadError)`? I applied your commands, but it didn't change anything.

MrSorcus, 2025-07-19:
> ```
> linking miniruby
> /bin/sh ./tool/ifchange "--timestamp=.rbconfig.time" rbconfig.rb rbconfig.tmp
> rbconfig.rb updated
> generating encdb.h
> ./tool/generic_erb.rb:6:in 'Kernel#require': cannot load [...]

Fixed by installing the `ruby-erb` package on the host.