openwrt/packages #25055: prometheus-node-exporter-lua-uci_dhcp_host crashes when mac address is a list

macjl, 2024-09-29:
Hello,

The file /usr/lib/lua/prometheus-collectors/uci_dhcp_host.lua doesn’t seems to accept a list of mac address.

In /etc/config/dhcp : 
```
config host
        option name 'myname'
        option ip '192.168.1.1'
        option dns '1'
        option mac 'AA:BB:CC:DD:EE:FF'
```
==> Works

But : 
```
config host
        option name 'myname'
        option ip '192.168.1.1'
        option dns '1'
        list mac 'AA:BB:CC:DD:EE:FF'
```
==> Crash

rogovogor, 2025-02-02:
+ confirm

djarosz, 2025-10-25:
Had same problem. For now I have changed _/usr/lib/lua/prometheus-collectors/uci_dhcp_host.lua _  locally 

from
```lua
local uci=require("uci")

local function scrape()
  local curs=uci.cursor()
  local metric_uci_host = metric("uci_dhcp_host", "gauge")

  curs:foreach("dhcp", "host", function(s)
    if s[".type"] == "host" then
      labels = {name=s["name"], mac=string.upper(s["mac"]), dns=s["dns"], ip=s["ip"]}
      metric_uci_host(labels, 1)
    end
  end)
end
```

to

```lua
local uci=require("uci")

local function scrape()
  local curs=uci.cursor()
  local metric_uci_host = metric("uci_dhcp_host", "gauge")

  curs:foreach("dhcp", "host", function(s)
    if s[".type"] == "host" then
      if type(s["mac"]) == "string" then
        labels = {name=s["name"], mac=string.upper(s["mac"]), dns=s["dns"], ip=s["ip"]}
      else
        labels = {name=s["name"], mac=string.upper(s["mac"][1]), dns=s["dns"], ip=s["ip"]}
      end
      metric_uci_host(labels, 1)
    end
  end)
end
```