openwrt/packages #25059: nginx-util: build failed with APK enabled

hingbong, 2024-09-30:
Description:
After enabling apk instead of okpg, `nginx-ssl-util` build failed
```
2024-09-30T14:58:11.1621363Z /home/runner/work/ax3600-builder/ax3600-builder/openwrt/staging_dir/host/bin/fakeroot /home/runner/work/ax3600-builder/ax3600-builder/openwrt/staging_dir/host/bin/apk mkpkg --info "name:nginx-ssl-util" --info "version:1.7-r1" --info "description:" --info "arch:aarch64_cortex-a53" --info "license:" --info "origin:feeds/packages/net/nginx-util" --info "provides:" --script "post-install:/home/runner/work/ax3600-builder/ax3600-builder/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-util-1.7/apk-aarch64_cortex-a53/nginx-ssl-util/post-install" --script "pre-deinstall:/home/runner/work/ax3600-builder/ax3600-builder/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-util-1.7/apk-aarch64_cortex-a53/nginx-ssl-util/pre-deinstall" --info "depends:nginx-util>=1.4-2 libc libstdcpp6 libuci20130104 libubus20231128 libubox20240329 libpthread libopenssl3 nginx-util libpcre2" --files "/home/runner/work/ax3600-builder/ax3600-builder/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-util-1.7/ipkg-aarch64_cortex-a53/nginx-ssl-util" --output "/home/runner/work/ax3600-builder/ax3600-builder/openwrt/bin/packages/aarch64_cortex-a53/packages/nginx-ssl-util-1.7-r1.apk" --sign "/home/runner/work/ax3600-builder/ax3600-builder/openwrt/private-key.pem"
2024-09-30T14:58:11.1740843Z ERROR: info field 'depends' has invalid value: dependency format is invalid
2024-09-30T14:58:11.1743206Z ERROR: failed to create package: /home/runner/work/ax3600-builder/ax3600-builder/openwrt/bin/packages/aarch64_cortex-a53/packages/nginx-ssl-util-1.7-r1.apk: dependency format is invalid
```

peter-stadler, 2024-10-03:
Thank you for the notification. For me it could take until December that I can look into it. So, if you have a fix, too, feel free to create a PR.

tiagogaspar8, 2024-10-26:
Adding to this, nginx seems to be having the same issue:
```
ERROR: info field 'depends' has invalid value: dependency format is invalid
ERROR: failed to create package: /mnt/openwrt/ss_mod/bin/packages/aarch64_cortex-a53/packages/nginx-ssl-1.26.1-r2.apk: dependency format is invalid
```
From my troubleshooting, it seems it has something to do with the versioning in the depends section.
Tagging @Ansuel

JasonD1980, 2024-11-13:
I had just finally compiled openwrt x86_64 for the very first time the other day, apparently that same day they removed all of the IPK packages.  So i enabled APK and tried to compile it again but i am getting this same dependency format is invalid error.  Has anyone figured out a proper fix for it yet?

tiagogaspar8, 2024-11-13:
You just have to remove the version dependency in the nginx makefile

Ansuel, 2024-11-13:
nginx-util has been dropped and fix has been pushed to master

closed 2024-11-13 by Ansuel

JasonD1980, 2024-11-14:
well i updated the source files, the correct version in the two areas as noted in ea5536c match the patch and it still failed compiling with same error, so i changed it to apktls instead of oppenssl.

`install -m0755 /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/nginx-ssl-util /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/.pkgdir/nginx-ssl-util/usr/bin/nginx-util
touch /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/.pkgdir/nginx-ssl-util.installed
mkdir -p /home/jason/Downloads/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/jason/Downloads/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/.pkgdir/nginx-ssl-util/. /home/jason/Downloads/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/jason/Downloads/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.nginx-ssl-util_installed
rm -rf /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util
mkdir -p /home/jason/Downloads/openwrt/bin/targets/x86/64/packages /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util /home/jason/Downloads/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/usr/bin
find /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/jason/Downloads/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/jason/Downloads/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/jason/Downloads/openwrt/staging_dir/host/bin/patchelf" /home/jason/Downloads/openwrt/scripts/rstrip.sh /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util
install -d -m0755 /home/jason/Downloads/openwrt/bin/packages/x86_64/packages/tmp
mkdir -p /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/
mkdir -p /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/
(cd /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util; )
( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo 'export root="${IPKG_INSTROOT}"'; echo 'export pkgname="nginx-util"'; echo "add_group_and_user"; [ ! -f /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/postinst-pkg ] || cat "/home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/postinst-pkg"; echo "default_postinst"; ) > /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/post-install;
( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo 'export root="${IPKG_INSTROOT}"'; echo 'export pkgname="nginx-util"'; [ ! -f /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/prerm-pkg ] || cat "/home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/prerm-pkg"; echo "default_prerm"; ) > /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/pre-deinstall;
if [ -n "" ]; then echo  > /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/nginx-util.rusers; fi;
if [ -n "" ]; then echo  > /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/nginx-util.alternatives; fi;
(cd /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util && find . -type f,l -printf "/%P\n" > /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/nginx-util.list)
# Move conffiles to IDIR and build conffiles_static with csums
if [ -f /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/conffiles ]; then mv -f /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/apk-all/nginx-util/conffiles /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/nginx-util.conffiles; for file in $(cat /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/nginx-util.conffiles); do [ -f /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/$file ] || continue; csum=$(/home/jason/Downloads/openwrt/staging_dir/host/bin/mkhash sha256 /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/$file); echo $file $csum >> /home/jason/Downloads/openwrt/build_dir/target-x86_64_musl/nginx-util-1.7/ipkg-all/nginx-util/lib/apk/packages/nginx-util.conffiles_static; done; fi
# Some package (base-files) manually append stuff to conffiles
# Append stuff from it and delete the CONTROL directory since everything else should be migrated
[... cut ...]
time: package/feeds/packages/nginx-util/compile#38.15#3.47#58.20
    ERROR: package/feeds/packages/nginx-util failed to build.
make[2]: *** [package/Makefile:184: package/feeds/packages/nginx-util/compile] Error 1
make[2]: Leaving directory '/home/jason/Downloads/openwrt'
make[1]: *** [package/Makefile:178: /home/jason/Downloads/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jason/Downloads/openwrt'
make: *** [/home/jason/Downloads/openwrt/include/toplevel.mk:248: world] Error 2
`