openwrt/packages #25061: ddns-scripts-digitalocean Not updating properly

thystra, 2024-09-30:
Maintainer: unable to locate do script repo
Environment:GL-AXT-1800 router, ARMv7, OpenWrt 21.02-SNAPSHOT r16399+172-c67509efd7 / LuCI openwrt-22.03 branch git-21.284.67084-e4d24f0

Description:
Package ddns-scripts-digitalocean version 2.8.2-12

Inputting values into the interface on LuCI does not result in a successful update to Digital Ocean IPV6. 

```
 155623       : parsing script '/usr/lib/ddns/update_digitalocean_com_v2.sh'
 155623       : DDNS Provider answered:
<html><head><title>Current IP Check</title></head><body>Current IP Address: 2605:59c8:5300:245:9683:c4ff:fe48:622c</body></html>
 155623 ERROR : IP update not accepted by DDNS Provider
 155623       : Waiting 600 seconds (Check Interval)
```
However, I'm able to achieve a successful update via commandline:
```
curl -X PUT -H "Content-Type: application/json" -H "Authorization: Bearer dop_v1_RESTOFAPIKEY" -d '{"name": "SUBDOMAIN-HOST-ONLY","data":"IPV6-ADDRESS-VALUE"}' "https://api.d
igitalocean.com/v2/domains/DOMAIN.TLD/records/DOMAINID" 
```

 DO Response:
```
 {"domain_record":{"id":DOMAINID,"type":"AAAA","name":"SUBDOMAIN-HOST-ONLY","data":"IPV6-ADDRESS-VALUE","priority":null,"port":null,"ttl":3600,"weight":null,"flags":null,"tag":null}}
```

Neustradamus, 2024-12-09:
@thystra: Have you tested the @feiming patch?
It works?

@feiming: It can not be merged because it is has been not done :/
Can you solve it?

closed 2025-02-04 by feckert

Neustradamus, 2025-02-04:
Thanks @Subcursion for this PR:
- https://github.com/openwrt/packages/pull/25888

PR #25888 (2025-02-04): ddns-scripts: add type field for digitalocean API
Maintainer: me
Compile tested: Compile not tested
Run tested: Architecture: ARMv7 Processor rev 0 (v7l), Platform: ipq806x/generic, Model: Netgear Nighthawk XR500

Description:
DigitalOcean API requires a "type" JSON field to update a DNS record. This adds that while checking for IPv6 to change which record type to use. Without it, the API call fails, making the script unable to update the DNS records for DigitalOcean.

Fixes #25061