openwrt/packages #25064: wifischedule - turns wifi off after power loss even if it is not time

Mile-Lile, 2024-10-01:
@rindeal 

wifischedule - turns wifi off after power loss even if it is not time

***

## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Services→ WiFI Schedule → Enable Wifi Schedule → Enable logging → Enable BUSINESSHOURS
2. Enable all days → Start wifi 04:00 → Stop wifi 00:00 → Enable Force disabling wifi even if stations associated
3. Save & Apply
4. reboot router (simulate power off) in e.g. 17:00

## Actual behavior:
**Router starts with wifi off even if it is BUSINESSHOURS when wifi should be on**

## Expected behavior:
**Wifi should be on (regardless of rebooting), and should be off at the midhinght, script should check clock somehow**

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r27598-603a3c6955'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='ARMv8 Processor rev 4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r27598-603a3c6955 / LuCI Master 24.264.56398~8449716'
ROUTER_MODEL='Xiaomi Mi Router AX3000T (OpenWrt U-Boot layout)'
```

and here are some errors from my logs, look at 19:19:44

[image]

systemcrash, 2024-10-01:
@rindeal - wonder whether you could also tweak the scripts to ensure they run after NTP time syncs? To exclude the possibility that a device with skewed clock doesn't make changes until it's synched to a realistic time source.

On my system, sysntpd runs with prio 98. So this should run at at least 99.

systemcrash, 2024-10-07:
Hi @rindeal - is there a fix in for this?

Mile-Lile, 2024-10-14:
I tested it in the way @rindeal suggested and my problem was solved even after router rebooting couple of times to make sure everything is working... So, can someone push this to snapshots master?

[image]

 Oct 14 17:21:14 AX3000T dropbear Exit (root) from <192.168.1.223:57048>: Disconnect received
 Oct 14 17:21:20 AX3000T procd - shutdown -
 Oct 14 17:21:20 AX3000T hostapd Remove interface 'phy1'
 Oct 14 17:21:20 AX3000T hostapd phy1-ap0: interface state ENABLED->DISABLED
 Oct 14 17:21:20 AX3000T hostapd phy1-ap0: AP-STA-DISCONNECTED 94:be:46:8c:a7:e7
 Oct 14 17:21:20 AX3000T hostapd phy1-ap0: AP-STA-DISCONNECTED 40:f3:08:ee:a5:28
 Oct 14 17:21:20 AX3000T hostapd phy1-ap0: AP-DISABLED
 Oct 14 17:21:20 AX3000T hostapd phy1-ap0: CTRL-EVENT-TERMINATING
 Oct 14 17:21:20 AX3000T hostapd rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
 Oct 14 17:21:20 AX3000T hostapd nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
 Oct 14 17:21:20 AX3000T dropbear Early exit: Terminated by signal
 Oct 14 17:21:20 AX3000T kernel [123591.409841] mt798x-wmac 18000000.wifi phy1-ap0: left allmulticast mode
 Oct 14 17:21:20 AX3000T kernel [123591.416805] mt798x-wmac 18000000.wifi phy1-ap0: left promiscuous mode
 Oct 14 17:21:20 AX3000T kernel [123591.423774] br-lan: port 5(phy1-ap0) entered disabled state
 Oct 14 17:21:20 AX3000T netifd Network device 'phy1-ap0' link is down
 Oct 14 17:21:56 AX3000T nextdns Connected 45.90.28.0:443 (con=35ms tls=101ms, TCP, TLS13)
 Oct 14 17:21:56 AX3000T nextdns Connected 194.71.126.128:443 (con=9ms tls=47ms, TCP, TLS13)
 Oct 14 17:21:56 AX3000T nextdns Switching endpoint: https://dns.nextdns.io#194.71.126.128,217.146.8.49,2a03:f80:381:194:71:126:128:1,2605:380:40:4::1
 Oct 14 17:21:56 AX3000T logread Logread connected to logs.papertrailapp.com:59656 via udp
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: STA 80:91:33:ee:92:79 IEEE 802.11: authenticated
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: STA 80:91:33:ee:92:79 IEEE 802.11: associated (aid 1)
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: AP-STA-CONNECTED 80:91:33:ee:92:79 auth_alg=open
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: STA 80:91:33:ee:92:79 WPA: pairwise key handshake completed (RSN)
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: EAPOL-4WAY-HS-COMPLETED 80:91:33:ee:92:79
 Oct 14 17:21:58 AX3000T dnsmasq-dhcp DHCPREQUEST(br-lan) 192.168.1.223 80:91:33:ee:92:79
 Oct 14 17:21:58 AX3000T dnsmasq-dhcp DHCPACK(br-lan) 192.168.1.223 80:91:33:ee:92:79 hp-laptop
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: STA 94:be:46:8c:a7:e7 IEEE 802.11: authenticated
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: STA 94:be:46:8c:a7:e7 IEEE 802.11: associated (aid 2)
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: AP-STA-CONNECTED 94:be:46:8c:a7:e7 auth_alg=open
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: STA 94:be:46:8c:a7:e7 WPA: pairwise key handshake completed (RSN)
 Oct 14 17:21:58 AX3000T hostapd phy0-ap0: EAPOL-4WAY-HS-COMPLETED 94:be:46:8c:a7:e7
 Oct 14 17:21:58 AX3000T dnsmasq-dhcp DHCPREQUEST(br-lan) 192.168.1.246 94:be:46:8c:a7:e7
 Oct 14 17:21:58 AX3000T dnsmasq-dhcp DHCPACK(br-lan) 192.168.1.246 94:be:46:8c:a7:e7
 Oct 14 17:22:04 AX3000T hostapd phy1-ap0: ACS-COMPLETED freq=5180 channel=36
 Oct 14 17:22:04 AX3000T hostapd phy1-ap0: interface state ACS->HT_SCAN
 Oct 14 17:22:05 AX3000T kernel [   31.445626] br-lan: port 5(phy1-ap0) entered blocking state
 Oct 14 17:22:05 AX3000T kernel [   31.451471] br-lan: port 5(phy1-ap0) entered forwarding state
 Oct 14 17:22:05 AX3000T netifd Network device 'phy1-ap0' link is up
 Oct 14 17:22:05 AX3000T hostapd phy1-ap0: interface state HT_SCAN->ENABLED
 Oct 14 17:22:05 AX3000T hostapd phy1-ap0: AP-ENABLED
 Oct 14 17:22:07 AX3000T banIP-1.0.0-r8 start banIP processing (boot)
 Oct 14 17:22:07 AX3000T hostapd phy0-ap0: STA 44:55:c4:eb:14:7c IEEE 802.11: authenticated
 Oct 14 17:22:07 AX3000T hostapd phy0-ap0: STA 44:55:c4:eb:14:7c IEEE 802.11: associated (aid 3)
 Oct 14 17:22:07 AX3000T banIP-1.0.0-r8 initialize banIP nftables namespace
 Oct 14 17:22:07 AX3000T hostapd phy0-ap0: AP-STA-CONNECTED 44:55:c4:eb:14:7c auth_alg=open
 Oct 14 17:22:07 AX3000T hostapd phy0-ap0: STA 44:55:c4:eb:14:7c WPA: pairwise key handshake completed (RSN)
 Oct 14 17:22:07 AX3000T hostapd phy0-ap0: EAPOL-4WAY-HS-COMPLETED 44:55:c4:eb:14:7c
 Oct 14 17:22:07 AX3000T banIP-1.0.0-r8 start banIP download processes
 Oct 14 17:22:07 AX3000T dnsmasq-dhcp DHCPREQUEST(br-lan) 192.168.1.212 44:55:c4:eb:14:7c
 Oct 14 17:22:07 AX3000T dnsmasq-dhcp DHCPACK(br-lan) 192.168.1.212 44:55:c4:eb:14:7c HONOR_8S-872ceb2025eb2d52
 Oct 14 17:22:07 AX3000T dnsmasq-dhcp DHCPREQUEST(br-lan) 192.168.1.212 44:55:c4:eb:14:7c
 Oct 14 17:22:07 AX3000T dnsmasq-dhcp DHCPACK(br-lan) 192.168.1.212 44:55:c4:eb:14:7c HONOR_8S-872ceb2025eb2d52
 Oct 14 17:22:09 AX3000T banIP-1.0.0-r8 start banIP domain lookup
 Oct 14 17:22:09 AX3000T banIP-1.0.0-r8 domain lookup finished in 0m 0s (allowlist, 0 domains, 0 IPs)
 Oct 14 17:22:09 AX3000T banIP-1.0.0-r8 domain lookup finished in 0m 0s (blocklist, 0 domains, 0 IPs)
 Oct 14 17:22:09 AX3000T banIP-1.0.0-r8 start detached banIP log service (/sbin/logread)
 Oct 14 17:22:24 AX3000T hostapd phy1-ap0: STA 40:f3:08:ee:a5:28 IEEE 802.11: authenticated
 Oct 14 17:22:24 AX3000T hostapd phy1-ap0: STA 40:f3:08:ee:a5:28 IEEE 802.11: associated (aid 1)
 Oct 14 17:22:25 AX3000T hostapd phy1-ap0: AP-STA-CONNECTED 40:f3:08:ee:a5:28 auth_alg=open
[... cut ...]
 Oct 14 17:23:02 AX3000T hostapd phy0-ap0: STA 94:be:46:8c:a8:79 WPA: pairwise key handshake completed (RSN)
 Oct 14 17:23:02 AX3000T hostapd phy0-ap0: EAPOL-4WAY-HS-COMPLETED 94:be:46:8c:a8:79
 Oct 14 17:23:02 AX3000T dnsmasq-dhcp DHCPREQUEST(br-lan) 192.168.1.185 94:be:46:8c:a8:79
 Oct 14 17:23:02 AX3000T dnsmasq-dhcp DHCPACK(br-lan) 192.168.1.185 94:be:46:8c:a8:79
 Oct 14 17:28:23 AX3000T dropbear Child connection from 192.168.1.223:57302
 Oct 14 17:28:26 AX3000T dropbear Password auth succeeded for 'root' from 192.168.1.223:57302

closed 2024-10-17 by hnyman