openwrt/packages #25070: [Request | udp2raw] New package

lars18th, 2024-10-02:
Maintainer: @codemarauder (Nishant Sharma <nishant@hopbox.in>), Toni Uhlig <matzeton@googlemail.com>  
(maintainers of the friend package `UDPspeeder`)

Description:

Add a package for `udp2raw` https://github.com/wangyu-/udp2raw

With this tool you can add to UDPspeeder the features of FakeTCP and ICMP transports.

Thank you.

brada4, 2024-10-05:
PR welcome?

BKPepe, 2024-10-18:
Yeah, if you want to have new packages in this repository, please create pull request.

closed 2024-10-18 by BKPepe

lars18th, 2024-10-18:
Hi @BKPepe ,

The idea to post this request here is because the _maintainers_ of the **friend** package `UDPspeeder` could create the PR. It's a simple call to them to add the linked `udp2raw` package.

Excuse me if this is against the rules of this forum.

brada4, 2024-10-18:
Take an inch of courage to become a co-maintainer.