openwrt/packages #25079: python-uci: Update to 0.10.0

Nemirtingas, 2024-10-03:
Maintainer: @BKPepe 

Description:
Update to 0.10.0 (allows creation of anonymous config) https://gitlab.nic.cz/turris/pyuci/-/tags/v0.10.0

BKPepe, 2024-10-03:
Thanks for creating such an issue, but I haven't seen that Turris has been using that version since it was released, and that's for more than one year already. Most likely, Turris package managers such as @muzikr or @Zatharalex should look into it. It's their package, after all, and they use it daily on their routers.

Nemirtingas, 2024-10-03:
Hi,
Any tips on how I could update it in my openwrt image? At my work here, they build their own openwrt image with embeded packages and kernel modules. I'm creating an ansible python module to manipulate uci configuration through python-uci.

I've updated the Makefile of feeds/packages/lang/python/python-uci like so:
```
#
# Copyright (C) 2018-2022 CZ.NIC, z. s. p. o. (https://www.nic.cz/)
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=python-uci
PKG_VERSION:=0.10.0
PKG_RELEASE:=1
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://gitlab.nic.cz/turris/pyuci
PKG_SOURCE_DATE:=2023-08-03
PKG_SOURCE_VERSION:=89c638444207277cab8ac76c302c18df9775d1e2
PKG_MIRROR_HASH:=5806d2ff9001c6286e8ae7a89cc506eda62d749d147be227145f60744fc2e947

#PYPI_NAME:=pyuci
#PKG_HASH:=847b8f365d1b5ff5e241cd0a666dff23098be27021b5aca982a48235dc4aeca2

PKG_MAINTAINER:=Josef Schlehofer <pepe.schlehofer@gmail.com>
PKG_LICENSE:=GPL-3.0-or-later
PKG_LICENSE_FILES:=LICENSE

include ../pypi.mk
include $(INCLUDE_DIR)/package.mk
include ../python3-package.mk

define Package/python3-uci
  SECTION:=lang
  CATEGORY:=Languages
  SUBMENU:=Python
  TITLE:=Python UCI bindings
  URL:=https://gitlab.nic.cz/turris/pyuci/
  DEPENDS:=+python3-light +libuci
endef

define Package/python3-uci/description
  Python3 bindings for Unified Configuration Interface.
endef

$(eval $(call Py3Package,python3-uci))
$(eval $(call BuildPackage,python3-uci))
$(eval $(call BuildPackage,python3-uci-src))
```

It seems to build fine but fails with the following error: `No such file or directory: '/workdir/openwrt/build_dir/target-x86_64_musl/python-uci-0.10.0//openwrt-build/python_uci-0.10.0-*.whl'`
The file in that directory is named `pyuci-0.10.0-cp311-cp311-linux_x86_64.whl`

Any idea?

BKPepe, 2024-10-03:
For the beginning, I think @shenek should upload new version of PyUCI to PyPI, because there isn

Nemirtingas, 2024-10-03:
Fixed my compile issue, replace `PKG_NAME:=python-uci`by `PKG_NAME:=pyuci`, that'll be my workaround until the package is officially updated.

Carmezim, 2025-02-03:
Any plans to release v0.10.0 on PyPI?

shenek, 2025-02-20:
Hi, version 0.10.2 should be available in PyPI.

shenek, 2025-02-27:
Note that pyuci with another bugfix was released. So the current version is `0.10.3` now.

Nemirtingas, 2025-02-27:
Hi, thank you for the updates. I noticed that the anonymous section are quite buggy. If the uci configuration has been loaded, a call to `.add` will throw an "Duplicate Entry" exception, section too has an issue, you have to create a new section before applying any change to it or you will encounter "Duplicate Entry" or "Bad argument" exceptions.

For example, I create an anonymous section, if I rename it and then try to change its content, it throws an exception. But if you create the anonymous section, you update it and then rename it, there is no exception.

Nemirtingas, 2025-02-27:
If you got some time, I'd be really grateful if you could implement python_ubus notify and subscribe. There is already old tickets about that on the gitlab, but I cannot connect to it and know very little about Python bindings.

BKPepe, 2025-05-21:
Guys, @muzikr and @Zatharalex, are you interested to update it here, so you can use it in your distribution or not? If not, we might remove not maintained packages.

muzikr, 2025-05-22:
I missed this. Sorry about that. I will update it here as soon as possible.

BKPepe, 2025-06-11:
Kind reminder. :)

muzikr, 2025-07-01:
I haven't forgotten about this, but I am struggling with it. The problem is that in newer versions, @shenek used `ext-modules` from setuptools (see [here](https://gitlab.nic.cz/turris/pyuci/-/blob/master/pyproject.toml?ref_type=heads#L29)). This feature was added to setuptools in v74.1.0 (see [here](https://setuptools.pypa.io/en/stable/history.html#v74-1-0)).

If I understand correctly, the version used for the build on the host is the one specified in [python3-version.mk](https://github.com/openwrt/packages/blob/c94ab433caf345ebc6bef9a7174d1f65f676fc87/lang/python/python3-version.mk#L18). Updating it would result in build failures for many packages.

I am still trying to find some solution.

BKPepe, 2025-07-14:
I am thinking a little bit loud, but what is wrong to update Python in this repository? 🤔 It would solve your problem. Yeah, there might be some incompatibility, but that's can be fixed as well, right? 😇