openwrt/packages #25089: PBR: No longer works after reinstalling OpenWRT 23.05.5

alehofde, 2024-10-06:
After reinstalling OpenWRT 23.05.5 with PBR 1.1.6-20, the service no longer starts

ERROR: The pbr 1.1.6-20 service failed to discover WAN gateway!

I still have an OpenWRT 23.05.03 running with pbr 1.1.1-7 and everything works fine there.

I'm sure it's a configuration error on my end but I have no idea :-(

Can anyone help?

alehofde, 2024-10-06:
[image]

I missing the service gateways. These are not displayed

stangri, 2024-10-06:
If you have updated the luci app or if you're seeing the error in the console, it usually includes the relevant link to the README. Follow it. If you already know it's not a bug but a misconfiguration, please post in the forum thread linked from README.

closed 2024-10-06 by stangri

spearmootz, 2024-10-08:
this same thing happened to me.  basically my router is connected to the internet by connecting to another router by WIFI.  the service never starts because its waiting for wan.  once i plug it in the service starts no problem.

any idea how this can be mitigated?  @stangri

stangri, 2024-10-08:
@spearmootz please post the CLI or luci app error message in its entirety.

spearmootz, 2024-10-08:
@stangri there is no error.  you click start and it just stays like the following image. then the logs get filled with waiting for WAN.  once you connect the internet cable it successfully starts.
[image]
[image]
[image]

spearmootz, 2024-10-08:
i am running on PRB Version 1.1.6-22 - Running (fw4 nft file mode).

stangri, 2024-10-08:
This looks like an outdated luci app. You need to either update it or clear the browser cache (or both).

You may benefit from reviewing the README and either use an option to not wait for wan on start or set the procd_wan_interface option to wwan (or both).

spearmootz, 2024-10-09:
i dont think i am able to update, i opened in incognito and got that version.  1.1.6-22 seems to be the latest on my version according to the opkg. 

setting the procd_wan_interface did work! thanks

amzamzamzamz, 2024-10-11:
the same issue with tor interface

Laitinlok, 2024-12-20:
I can no longer choose AdguardHome ipset with the new version of pbf-iptables.

stangri, 2024-12-20:
> I can no longer choose AdguardHome ipset with the new version of pbr -iptables.

Can you please post the output of: `ubus -S call luci.pbr getPlatformSupport '{"name": "pbr" }'`?