openwrt/packages #25109: tc-full tc-tiny ipk file made for 4.19 kernel request

NoUserNameForYou, 2024-10-09:
Hello,

My archer c7 v1 has 5ghz support with a custom firmware made by one of your members, but I can't install sqm because tc isn't compatible with its kernel which is 4.19. I see the kernels were 4.14 and jumps to 5.4 closest in your official releases.

Firmware has its luci web software rep directed to the snapshots packages, I edited distro config files and added 19.07, 18.06 etc. and pulled old versions but none of the ipks target 4.19.

the build itself, https://github.com/d687r02j8g/openwrt_c7v1/raw/master/ath10k.zip and related discussion topic with more info https://forum.openwrt.org/t/archer-c7-v1-0-crashes-on-boot/19784/ 

Could someone be kind enough to direct me where I can get those supported tc files so I can install sqm luci app? 

Thank you

feckert, 2024-10-09:
The release 18.06 is EOL and schould not be used.
If you still want to download prebuild package you must know the architecture and the openwrt release.
For example openwrt point release 18.06.8 (architecture x86_64) you can download ip-tiny from the following URL https://archive.openwrt.org/releases/18.06.9/packages/x86_64/base/ip-tiny_4.16.0-4_x86_64.ipk

closed 2024-10-09 by feckert

NoUserNameForYou, 2024-10-09:
Yes, that's what I did. I added the 18.06/19.07 etc repos, but all the TC files target either 4.14 or 5.4 kernels as I've mentioned in my initial post. I'm looking for a tc build from  2020/2019 times snapshots. Are those even saved?

See "2020-10-30 	Adrian Schmutzler	kernel: remove support for kernel 4.19"
https://git.openwrt.org/?p=openwrt/openwrt.git;a=history;f=target/linux/generic/hack-4.19/902-debloat_proc.patch;h=f337cd2ba377aed1e592828c638106083ace9caf

feckert, 2024-10-10:
I'm afraid you won't have any luck. As far as I know, the snapshot builds are only available in the Master branch!