openwrt/packages #25113: libevent2-core has wrong hash

ascendbeing, 2024-10-09:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

I'm building a custom openwrt build and it keeps falling at it says hash mismatch for libevent2-core. Which I apparently pulled in thanks to tmux.

This is likely why my builds kept failing. I don't know in which way it's broken but you get a different hash than you have listed 99/100+ times you try.

As soon as I removed it and tmux, I can now produce my builds. It's been like this for ~4 days at least. I tried updating the feeds apparently nobody noticed?

ascendbeing, 2024-10-09:
ping @jow-

neheb, 2024-10-30:
this belongs in openwrt/openwrt

GrumpyMeow, 2025-01-13:
Unable to reproduce issue on buildroot environment.
Both libevent2 and tmux download&compile fine. 
Can this issue be closed?

closed 2025-01-13 by neheb