openwrt/packages #25114: rust: Fails to compile the host Rust compiler configured for x86_64 and glibc when building in Ubuntu

aaronnevarez, 2024-10-10:
Maintainer: @lu-zero @krant 
Environment: x86_64 and glibc, OpenWrt 23.05. I tried this procedure using Ubuntu 20.04 and 22.04 LTS.

Description:
It seems on the build step tries to use the host glibc instead of the one built by OpenWrt.

To reproduce follow instructions from https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
In the make menu config select: x86_64 and glibc.

build using:
```
make -j$(nproc) defconfig download clean world
```
Once it successfully finished, build rust/host:
```
make -j1 V=s  package/rust/host/compile
```
and the following error occurs:
```
$ make -j1 V=s  package/rust/host/compile
make[2]: Entering directory '/home/anevarez/build/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/anevarez/build/openwrt/scripts/config'
make[1]: Entering directory '/home/anevarez/build/openwrt'
make[2]: Entering directory '/home/anevarez/build/openwrt/feeds/packages/lang/rust'
. /home/anevarez/build/openwrt/include/shell.sh; /home/anevarez/build/openwrt/staging_dir/host/bin/libdeflate-gzip -dc /home/anevarez/build/openwrt/dl/rustc-1.80.1-src.tar.gz | tar -C /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src

Applying /home/anevarez/build/openwrt/feeds/packages/lang/rust/patches/0001-Update-xz2-and-use-it-static.patch using plaintext: 
patching file src/bootstrap/Cargo.toml

Applying /home/anevarez/build/openwrt/feeds/packages/lang/rust/patches/0003-bump-libc-deps-to-0.2.146.patch using plaintext: 
patching file vendor/elasticlunr-rs-3.0.2/Cargo.lock
patching file vendor/libffi-3.2.0/Cargo.lock
touch /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/.prepared9388c6f52a846e342ef6ec61ca4ad92a_6664517399ebbbc92a37c5bb081b5c53
(cd /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/; if [ -x configure ]; then cp -fpR /home/anevarez/build/openwrt/scripts/config.{guess,sub} /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src// && CC="gcc" CFLAGS="-O2 -I/home/anevarez/build/openwrt/staging_dir/host/include -I/home/anevarez/build/openwrt/staging_dir/hostpkg/include -I/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/include" CXX="g++" CPPFLAGS="-I/home/anevarez/build/openwrt/staging_dir/host/include -I/home/anevarez/build/openwrt/staging_dir/hostpkg/include -I/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/include" CXXFLAGS="" LDFLAGS="-L/home/anevarez/build/openwrt/staging_dir/host/lib -L/home/anevarez/build/openwrt/staging_dir/hostpkg/lib -L/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/lib" CONFIG_SHELL="/usr/bin/env bash" CARGO_HOME="/home/anevarez/build/openwrt/dl/cargo"  bash ./configure --build=x86_64-unknown-linux-gnu --target=x86_64-unknown-linux-gnu,x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --prefix=/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host --bindir=/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/bin --libdir=/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/lib --sysconfdir=/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/etc --datadir=/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/share --mandir=/home/anevarez/build/openwrt/staging_dir/target-x86_64_glibc/host/man --dist-compression-formats=gz --disable-sanitizers --release-channel=stable --enable-cargo-native-static --bootstrap-cache-path=/home/anevarez/build/openwrt/dl/rustc --set=llvm.download-ci-llvm=true --set=target.x86_64-unknown-linux-gnu.ar=x86_64-openwrt-linux-gnu-gcc-ar --set=target.x86_64-unknown-linux-gnu.cc=x86_64-openwrt-linux-gnu-gcc --set=target.x86_64-unknown-linux-gnu.cxx=x86_64-openwrt-linux-gnu-g++ --set=target.x86_64-unknown-linux-gnu.linker=x86_64-openwrt-linux-gnu-gcc --set=target.x86_64-unknown-linux-gnu.ranlib=x86_64-openwrt-linux-gnu-gcc-ranlib --set=target.x86_64-unknown-linux-gnu.crt-static=false  ; fi )
configure: processing command line
configure: 
configure: build.configure-args := ['--build=x86_64-unknown-linux-gnu', '--target ...
configure: build.build          := x86_64-unknown-linux-gnu
configure: build.target         := ['x86_64-unknown-linux-gnu', 'x86_64-unknown-l ...
configure: build.host           := ['x86_64-unknown-linux-gnu']
configure: install.prefix       := /home/anevarez/build/openwrt/staging_dir/targe ...
configure: install.bindir       := /home/anevarez/build/openwrt/staging_dir/targe ...
configure: install.libdir       := /home/anevarez/build/openwrt/staging_dir/targe ...
configure: install.sysconfdir   := /home/anevarez/build/openwrt/staging_dir/targe ...
configure: install.datadir      := /home/anevarez/build/openwrt/staging_dir/targe ...
configure: install.mandir       := /home/anevarez/build/openwrt/staging_dir/targe ...
configure: dist.compression-formats := ['gz']
configure: build.sanitizers     := False
configure: rust.channel         := stable
configure: build.cargo-native-static := True
configure: build.bootstrap-cache-path := /home/anevarez/build/openwrt/dl/rustc
configure: llvm.download-ci-llvm := True
configure: target.x86_64-unknown-linux-gnu.ar := x86_64-openwrt-linux-gnu-gcc-ar
configure: target.x86_64-unknown-linux-gnu.cc := x86_64-openwrt-linux-gnu-gcc
configure: target.x86_64-unknown-linux-gnu.cxx := x86_64-openwrt-linux-gnu-g++
configure: target.x86_64-unknown-linux-gnu.linker := x86_64-openwrt-linux-gnu-gcc ...
configure: target.x86_64-unknown-linux-gnu.ranlib := x86_64-openwrt-linux-gnu-gcc ...
configure: target.x86_64-unknown-linux-gnu.crt-static := False
configure: profile              := dist
configure: 
configure: writing `config.toml` in current directory
configure: 
configure: run `python /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/x.py --help`
touch /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/.configured
CARGO_HOME=/home/anevarez/build/openwrt/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/rust-1.80.1=rust-1.80.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" python3 /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/x.py --build-dir /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/build --config /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
extracting /home/anevarez/build/openwrt/dl/rustc/2024-06-13/rust-std-1.79.0-x86_64-unknown-linux-gnu.tar.xz
extracting /home/anevarez/build/openwrt/dl/rustc/2024-06-13/rustc-1.79.0-x86_64-unknown-linux-gnu.tar.xz
extracting /home/anevarez/build/openwrt/dl/rustc/2024-06-13/cargo-1.79.0-x86_64-unknown-linux-gnu.tar.xz
[... 5 lines cut ...]
   Compiling version_check v0.9.4
   Compiling memchr v2.7.1
   Compiling libc v0.2.155
   Compiling rustix v0.38.28
   Compiling serde v1.0.195
   Compiling cc v1.0.97
error: failed to run custom build command for `proc-macro2 v1.0.76`

Caused by:
  process didn't exit successfully: `/home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/build/bootstrap/debug/build/proc-macro2-cfb5d0879a2ccf9b/build-script-build` (exit status: 1)
  --- stderr
  /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/build/bootstrap/debug/build/proc-macro2-cfb5d0879a2ccf9b/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/build/bootstrap/debug/build/proc-macro2-cfb5d0879a2ccf9b/build-script-build)
  /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/build/bootstrap/debug/build/proc-macro2-cfb5d0879a2ccf9b/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/build/bootstrap/debug/build/proc-macro2-cfb5d0879a2ccf9b/build-script-build)
[... cut ...]
make[2]: *** [Makefile:109: /home/anevarez/build/openwrt/build_dir/target-x86_64_glibc/host/rustc-1.80.1-src/.built] Error 1
make[2]: Leaving directory '/home/anevarez/build/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#29.43#28.05#43.34
    ERROR: package/feeds/packages/rust [host] failed to build.
make[1]: *** [package/Makefile:127: package/feeds/packages/rust/host/compile] Error 1
make[1]: Leaving directory '/home/anevarez/build/openwrt'
make: *** [/home/anevarez/build/openwrt/include/toplevel.mk:233: package/rust/host/compile] Error 2
```

lu-zero, 2024-10-10:
The build-script has to be built for the target, so it would be correct, but apparently rust targets a much newer glibc?, is it working on ubuntu 24?

aaronnevarez, 2024-10-10:
Hi @lu-zero,
Rust targets a much newer glibc, which appears to be in line with the toolchain built by OpenWrt. I have not tried it on Ubuntu 24. Ubuntu 22 gets a bit further than Ubuntu 20, but I would think it should run independently of the OS version, no?

It seems that when building for other arch/libc combinations, it uses the correct combination of libc/toolchain built by OpenWrt. However, when the arch/libc matches the one on the host, it gets a bit confused because it uses the toolchain built by OpenWrt but tries to use the libc from the host. It feels like the Makefile should set up the LD_LIBRARY_PATH and/or LIBRARY_PATH variables.

lu-zero, 2024-10-10:
It is quite a corner case, I'm not even sure there is a way to workaround it since for rustc the target is exactly the same.

aaronnevarez, 2024-10-10:
I have a workaround where I remove configuring the the pre-compiled toolchain so it uses the host toolchain to build which removes the mismatch between the toolchain and libc, with this I'm able to build the host Rust. However this is not the right solution as it would break all other arch/libc:
```
$ git diff lang/rust/Makefile
diff --git a/lang/rust/Makefile b/lang/rust/Makefile
index 78d95d3d9..94aa417ab 100644
--- a/lang/rust/Makefile
+++ b/lang/rust/Makefile
@@ -48,11 +48,6 @@ RUST_UNINSTALL:=$(STAGING_DIR)/host/lib/rustlib/uninstall.sh
 
 # Target Flags
 TARGET_CONFIGURE_ARGS = \
-       --set=target.$(RUSTC_TARGET_ARCH).ar=$(TARGET_AR) \
-       --set=target.$(RUSTC_TARGET_ARCH).cc=$(TARGET_CC_NOCACHE) \
-       --set=target.$(RUSTC_TARGET_ARCH).cxx=$(TARGET_CXX_NOCACHE) \
-       --set=target.$(RUSTC_TARGET_ARCH).linker=$(TARGET_CC_NOCACHE) \
-       --set=target.$(RUSTC_TARGET_ARCH).ranlib=$(TARGET_RANLIB) \
        --set=target.$(RUSTC_TARGET_ARCH).crt-static=false \
        $(if $(CONFIG_USE_MUSL),--set=target.$(RUSTC_TARGET_ARCH).musl-root=$(TOOLCHAIN_DIR))
```