openwrt/packages #25119: mwan3: Routes don't get removed from main routing table unless interface fully down

joshuadouch, 2024-10-10:
Maintainer: @feckert
Environment: OpenWrt 23.05.5, x86_64

Description:
I have configured mwan3, but when an interface loses connectivity (e.g. stops responding to ping) the routes don't get removed. If the interface is physically stopped, wait for failover and then started, then the routes will be removed from the main routing table. This is the case in both failover and recovery. 

I have configured this from scratch, ensuring I abide by all naming conventions, max character lengths and avoiding same name interface/member/policy/rules. However, I have also used the default example configuration (only changing config to match interface names), and I face the same issue. I have also tried this from a fresh installation of OpenWrt.

Is this a known issue? Is this package still maintained?

feckert, 2024-10-11:
The route is not deleted from the main route table. This is intentional. Whether this is good or bad is the question. But the mwan3 does it like this

In your case you have to change the metric of your default routings. The interface with the lowest metric wins here. And the traffic is then processed via this interface.

joshuadouch, 2024-10-11:
Thank you - I understand how this works better now. I was also getting confused by some unexpected behaviour from xl2tpd - it seemed to add a static route to the VPN endpoint automatically, and this was always over my backup WAN link, I assume because this always came online first at boot. 

To avoid this issue I have added the following to the Notify script

if [ "${ACTION}" = "disconnected" ] || [ "${ACTION}" = "connected" ] && [ "${INTERFACE}" = "wan_main" ] ; then
 ifdown aaisp
 ip route delete 90.155.xx.xx
 ifup aaisp
fi

And simply have two rules - one with the VPN endpoint at the top set to use a policy containing both main and backup links with a metric of 1 and 2 respectively, and a 'default' rule for everything else, routing everything over the VPN.

This was no issue with mwan3, apologies!

Thank you!

closed 2024-10-11 by joshuadouch

feckert, 2024-10-11:
> This was no issue with mwan3, apologies!

No problem :-)