openwrt/packages #25134: sstp-client: Update to 1.0.20

Neustradamus, 2024-10-13:
Maintainer: @fededim, @jow-, @hnyman, @champtar, @neheb, @harveyhu2012, @rkkoszewski, @lipnitsk, @paulfertser, @aparcar, @xback, @vgaetera
Environment: ALL

Description:
```
Update sstp-client to 1.0.20
```

Thanks in advance.

cc: @enaess

Links:
- https://github.com/openwrt/packages/tree/master/net/sstp-client
- https://gitlab.com/sstp-project/sstp-client/
- https://gitlab.com/sstp-project/sstp-client/-/releases
- https://sourceforge.net/projects/sstp-client/

rkkoszewski, 2024-10-15:
I can have a look at it.

enaess, 2024-10-15:
Let me know if you have any questions, @rkkoszewski

Neustradamus, 2024-11-03:
Thanks @Ansuel, he has done it!
- https://github.com/openwrt/packages/pull/25185

closed 2024-11-03 by Neustradamus