openwrt/packages #25163: fail2ban still depends in iptables

peci1, 2024-10-18:
Maintainer: @erdoukki 
Environment: git

Description:
fail2ban still [depends on iptables](https://github.com/openwrt/packages/blob/dd2e7105e6f0abf9b72155e2b26298ed19a92ab9/net/fail2ban/Makefile#L32) instead of firewall4.

It seems there are nftables configs installed with it, so it should be easy to switch the dependency. However, downstream users will need to manually re-specify their actions.

BKPepe, 2024-10-18:
Several packages, including fail2ban are still using iptables for various reasons:
- 1) The guy who added the package no longer use it / do not have time to look at it
In the case of fail2ban, a recent commit in this repository, which is relevant, is from last year, and upstream is releasing a new version.

**This could be a solution: someone from the community steps in as a volunteer tries to keep it up-to-date or sends a pull request to use nftables.**

- 2) Package is no longer maintained in upstream (not relevant for this package), downstream
Solution for this: rather remove those packages, which are out of date and not maintained in upstream.

This issue is somehow duplicated to https://github.com/openwrt/packages/issues/16818, where you can find a list of packages that are still using iptables, and as you can see, it is a quite a shortlist. Any help is appreciated.

closed 2024-10-18 by BKPepe

peci1, 2024-11-16:
I think the problem is just the declared dependency. fail2ban already provides nftables-based actions. So the only thing that is needed (AFAIK) is to drop the iptables dependency in OpenWRT. Is this a good way to proceed? Should I prepare a MR?