openwrt/packages #25166: prel gcc14 doio.c:218:9: error: implicit declaration of function 'dup3'; did you mean 'dup2'? [-Wimplicit-function-declaration]

joyplayer, 2024-10-19:
doio.c: In function 'Perl_PerlLIO_dup2_cloexec':
doio.c:218:9: error: implicit declaration of function 'dup3'; did you mean 'dup2'? [-Wimplicit-function-declaration]
  218 |         dup3(oldfd, newfd, O_CLOEXEC),
      |         ^~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:216:5: note: in expansion of macro 'DO_ONEOPEN_EXPERIMENTING_CLOEXEC'
  216 |     DO_ONEOPEN_EXPERIMENTING_CLOEXEC(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
doio.c: In function 'Perl_my_mkstemp_cloexec':
util.h:273:45: error: implicit declaration of function 'mkostemp'; did you mean 'mkstemp'? [-Wimplicit-function-declaration]
  273 | #   define Perl_my_mkostemp(templte, flags) mkostemp(templte, flags)
      |                                             ^~~~~~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:336:5: note: in expansion of macro 'DO_ONEOPEN_EXPERIMENTING_CLOEXEC'
  336 |     DO_ONEOPEN_EXPERIMENTING_CLOEXEC(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
doio.c:338:9: note: in expansion of macro 'Perl_my_mkostemp'
  338 |         Perl_my_mkostemp(templte, O_CLOEXEC),
      |         ^~~~~~~~~~~~~~~~
doio.c: In function 'Perl_PerlProc_pipe_cloexec':
doio.c:372:9: error: implicit declaration of function 'pipe2'; did you mean 'pipe'? [-Wimplicit-function-declaration]
  372 |         pipe2(pipefd, O_CLOEXEC),
      |         ^~~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:371:5: note: in expansion of macro 'DO_PIPEOPEN_EXPERIMENTING_CLOEXEC'
  371 |     DO_PIPEOPEN_EXPERIMENTING_CLOEXEC(PL_strategy_pipe, pipefd,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
doio.c: In function 'Perl_PerlSock_accept_cloexec':
doio.c:408:9: error: implicit declaration of function 'accept4'; did you mean 'accept'? [-Wimplicit-function-declaration]
  408 |         accept4(listenfd, addr, addrlen, SOCK_CLOEXEC),
      |         ^~~~~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:406:5: note: in expansion of macro 'DO_ONEOPEN_EXPERIMENTING_CLOEXEC'
  406 |     DO_ONEOPEN_EXPERIMENTING_CLOEXEC(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [makefile:266: doio.o] Error 1

joyplayer, 2024-10-20:
[perl.tar.gz](https://github.com/user-attachments/files/17448965/perl.tar.gz)
test it

neheb, 2024-10-27:
https://github.com/openwrt/packages/commit/ea8e6fbf33728ecc0546764b115bd2701d87e4aa is not enough?

catsimple, 2025-01-13:
```
make[5]: Leaving directory '/home/catsimple/openwrt/build_dir/target-x86_64_glibc/perl/perl-5.40.0'
Updating makefile...
make[4]: Leaving directory '/home/catsimple/openwrt/build_dir/target-x86_64_glibc/perl/perl-5.40.0'
make  -C /home/catsimple/openwrt/build_dir/target-x86_64_glibc/perl/perl-5.40.0
make[4]: Entering directory '/home/catsimple/openwrt/build_dir/target-x86_64_glibc/perl/perl-5.40.0'
x86_64-openwrt-linux-gnu-gcc -c -DPERL_CORE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -I/home/catsimple/openwrt/staging_dir/toolchain-x86_64_gcc-14.2.0_glibc/usr/include -I/home/catsimple/openwrt/staging_dir/toolchain-x86_64_gcc-14.2.0_glibc/include -O2 -Wall -Werror=pointer-arith -Werror=vla -Wextra -Wno-long-long -Wno-declaration-after-statement -Wc++-compat -Wwrite-strings -fPIC doio.c
doio.c: In function 'Perl_PerlLIO_dup2_cloexec':
doio.c:218:9: error: implicit declaration of function 'dup3'; did you mean 'dup2'? [-Wimplicit-function-declaration]
  218 |         dup3(oldfd, newfd, O_CLOEXEC),
      |         ^~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:216:5: note: in expansion of macro 'DO_ONEOPEN_EXPERIMENTING_CLOEXEC'
  216 |     DO_ONEOPEN_EXPERIMENTING_CLOEXEC(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
doio.c: In function 'Perl_my_mkstemp_cloexec':
util.h:275:45: error: implicit declaration of function 'mkostemp'; did you mean 'mkstemp'? [-Wimplicit-function-declaration]
  275 | #   define Perl_my_mkostemp(templte, flags) mkostemp(templte, flags)
      |                                             ^~~~~~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:336:5: note: in expansion of macro 'DO_ONEOPEN_EXPERIMENTING_CLOEXEC'
  336 |     DO_ONEOPEN_EXPERIMENTING_CLOEXEC(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
doio.c:338:9: note: in expansion of macro 'Perl_my_mkostemp'
  338 |         Perl_my_mkostemp(templte, O_CLOEXEC),
      |         ^~~~~~~~~~~~~~~~
doio.c: In function 'Perl_PerlProc_pipe_cloexec':
doio.c:372:9: error: implicit declaration of function 'pipe2'; did you mean 'pipe'? [-Wimplicit-function-declaration]
  372 |         pipe2(pipefd, O_CLOEXEC),
      |         ^~~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:371:5: note: in expansion of macro 'DO_PIPEOPEN_EXPERIMENTING_CLOEXEC'
  371 |     DO_PIPEOPEN_EXPERIMENTING_CLOEXEC(PL_strategy_pipe, pipefd,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
doio.c: In function 'Perl_PerlSock_accept_cloexec':
doio.c:408:9: error: implicit declaration of function 'accept4'; did you mean 'accept'? [-Wimplicit-function-declaration]
  408 |         accept4(listenfd, addr, addrlen, SOCK_CLOEXEC),
      |         ^~~~~~~
doio.c:125:32: note: in definition of macro 'DO_GENOPEN_EXPERIMENTING_CLOEXEC'
  125 |                     int res = (GENOPEN_CLOEXEC), eno; \
      |                                ^~~~~~~~~~~~~~~
doio.c:406:5: note: in expansion of macro 'DO_ONEOPEN_EXPERIMENTING_CLOEXEC'
  406 |     DO_ONEOPEN_EXPERIMENTING_CLOEXEC(
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[4]: *** [makefile:266: doio.o] Error 1
make[4]: Leaving directory '/home/catsimple/openwrt/build_dir/target-x86_64_glibc/perl/perl-5.40.0'
make[3]: *** [Makefile:151: /home/catsimple/openwrt/build_dir/target-x86_64_glibc/perl/perl-5.40.0/.built] Error 2
make[3]: Leaving directory '/home/catsimple/openwrt/feeds/packages/lang/perl'
time: package/feeds/packages/perl/compile#5.24#5.04#9.99
    ERROR: package/feeds/packages/perl failed to build.

```

Prototik, 2025-03-03:
Seems like glibc issue, with musl same config builds fine.

neheb, 2025-04-29:
make package/perl/{clean,compile}

yingziwu, 2025-05-01:
> make package/perl/{clean,compile}

still meeting the problem.

openwrt version: SNAPSHOT 2037ca29552eb8e158493c39b6922244d33334f2
target: generic x86

logs: [perl.tar.gz](https://github.com/user-attachments/files/19986905/perl.tar.gz)

schuettecarsten, 2025-08-24:
I have the same problem with GCC14 and GCC15.

rockdrilla, 2025-08-24:
There's deep-dive bug about host/target compilation: e.g. if some package requires "perl/host" when "perl" for target is built too despite not being selected in either (explicit/implicit) way. Furthermore, some packages will not build until you install "gcc-multilib" in host OS (in case of Linux).

schuettecarsten, 2025-10-22:
Any news on this? The workaround does not work any more...