openwrt/packages #25168: [bug] openwisp-config::instance1 s in a crash loop x

diepeterpan, 2024-10-19:
Maintainer: @pandafy, @nemesifier 

**Environment:** mediatek/filogic, Cudy WR3000 v1, OpenWrt 23.05.5 r24106-10cc5fcd00

But likely impacting all users of package **openwisp-config version 1.1.0-1**

**Description:**
Upgraded the package of openwisp-config to version 1.1.0-1 in OpenWrt 23.05.5. I couldn't connect to the controller anymore.

In logs this message appears (and controller doesn't see the client):

`Instance openwisp_config::instance1 s in a crash loop 6`

**Also see:**

[https://github.com/openwisp/openwisp-config/issues/223](https://github.com/openwisp/openwisp-config/issues/223)

Hunting down the bug I discovered that /etc/init.d/openwisp_config wants to execute PROG="/usr/sbin/openwisp-config" but it does not exists, it is called PROG="/usr/sbin/openwisp_config" according to the changelog there were changes re: the - 'hyphen' vs _ 'underscore' ; maybe it just gets packaged incorrectly? Once I edited /etc/init.d/openwisp_config and changed the - 'hyphen' to a _ 'underscore' in PROG= the new agent also started to work. Easier than rolling back.

Looks like the problem is in the OpenWrt Makefile - https://github.com/openwrt/packages/blob/dd2e7105e6f0abf9b72155e2b26298ed19a92ab9/admin/openwisp-config/Makefile which needs to be updated to produce hyphens instead of underscores.

```
	$(INSTALL_BIN) \
		$(PKG_BUILD_DIR)/openwisp-config/files/openwisp.agent \
		$(1)/usr/sbin/openwisp_config

	$(INSTALL_BIN) \
		$(PKG_BUILD_DIR)/openwisp-config/files/openwisp.init \
		$(1)/etc/init.d/openwisp_config
```

should maybe be

```
	$(INSTALL_BIN) \
		$(PKG_BUILD_DIR)/openwisp-config/files/openwisp.agent \
		$(1)/usr/sbin/openwisp-config

	$(INSTALL_BIN) \
		$(PKG_BUILD_DIR)/openwisp-config/files/openwisp.init \
		$(1)/etc/init.d/openwisp-config

```

nemesifier, 2024-10-22:
We'll send a fix asap.

nemesifier, 2024-10-22:
https://github.com/openwrt/packages/pull/25186 should fix.

PR #25186 (2024-11-11): openwisp-config: fix Makefile for 1.1.0
Update configuration in Makefile to fix #25168

Maintainer: @nemesifier 
Compile tested: ramips
Run tested: ramips
commit 67d4fa3b (2024-11-11): openwisp-config: fix Makefile for 1.1.0

Update configuration in Makefile to fix #25168.
Add "/etc/openwisp/" to conf files.
PR #25318 (2024-11-12): [24.10] openwisp-config: fix Makefile for 1.1.0
Maintainer: @nemesifier
Compile tested: ramips
Run tested: ramips

Description:

Update configuration in Makefile to fix #25168.
Add "/etc/openwisp/" to conf files.

(cherry picked from commit 67d4fa3b05e0a8ee0724bc245d6eec318351876a)
commit 15fe770c (2024-11-12): openwisp-config: fix Makefile for 1.1.0

Update configuration in Makefile to fix #25168.
Add "/etc/openwisp/" to conf files.

(cherry picked from commit 67d4fa3b05e0a8ee0724bc245d6eec318351876a)
closed 2024-11-11 by feckert