openwrt/packages #25172: sqm-scripts: not auto starting

uski, 2024-10-20:
Maintainer: @\tohojo
Environment: r24106-10cc5fcd00, 23.05.5, mediatek/filogic, GL.iNet GL-MT3000

Description:

After configuring sqm-scripts using luci-app-sqm, sqm works immediately.
However, after rebooting, sqm policies are no longer enforce and bandwidth is unlimited.
Running `/etc/init.d/sqm start` immediately restores the sqm policies.

Additionally, when setting the sqm loglevel to trace, the sqm logs are only visible when ran manually. There is no trace of sqm operation at boot time. This further supports that sqm is not auto-starting.

uski, 2024-10-20:
@tohojo

uski, 2024-10-20:
One noteworthy information is that in my setup, sqm is configured to operate only on a sub-interface (phy0-ap1). It is possible that there is a race condition at boot where sqm does not see this interface and does not execute, because the interface may take some time to be brought up at boot time. This is purely a conjecture however.

uski, 2024-10-20:
Adding the following to /etc/rc.local is a workaround that works for me:

`sh -c 'sleep 15 && /etc/init.d/sqm start' &`

tohojo, 2024-10-24:
Could you please post the contents of your /etc/config/sqm ?

uski, 2025-04-22:
Hi,

Had this issue again on another router with 24.10.0. Exactly the same scenario.

```
root@OpenWrt:~# cat /etc/config/sqm

config queue 'eth1'
	option enabled '1'
	option interface 'phy1-ap1'
	option download '6000'
	option upload '10000'
	option qdisc 'cake'
	option script 'piece_of_cake.qos'
	option linklayer 'none'
	option debug_logging '0'
	option verbosity '5'
```

This time I needed a delay of ~120 seconds. 15 seconds did not work.

This is consistent with OpenWRT trying to execute sqm init script ONCE, BEFORE the interface exists/is ready.
Which would be why executing the script manually after the interface had enough time to be brought up fixes the issue. This is because this is a softAP interface, not a hardwired interface (I want to throttle the bandwidth of one specific Wifi network, which I assume is a common use case for guest networks etc.)

OpenWRT should re-configure SQM every time there is a change in interfaces.

tohojo, 2025-04-22:
uski ***@***.***> writes:

> OpenWRT should re-configure SQM every time there is a change in
> interfaces.

It already does, by way of a hotplug script
(/etc/hotplug.d/iface/11-sqm).

So if your settings are not being applied, either the hotplug script is
not being executed at all for WiFi interfaces, or the logic in
sqm-script's interface-to-config mapping is not working properly for
WiFi interfaces. I suspect the latter.

Could you please add a debug hotplug script to show which events are
being emitted while setting up the interface? Something like (in /etc/hotplug/iface/00-debug.sh):

#!/bin/sh

echo $(date) $INTERFACE $DEVICE $ACTION >> /tmp/hotplug.log

Then post the output of /tmp/hotplug.log after re-configuring your WiFi
interface?

gdluca, 2026-01-19:
@tohojo 

Hi, I think I have the same issue and I can answer your request.
I put that debug hotplug script in /etc/hotplug.d/iface/00-debug.sh and made a change in my wireless config but sqm did not get applied on iface "phy1-ap0". Also no /tmp/hotplug.log was created so something is failing somewhere.
To restore sqm, I need to disable the sqm instance on Luci and then enable it again.