openwrt/packages #25175: tailscale: missing mips64 support + security vulnerability

uski, 2024-10-20:
Maintainer: @mochaaP @supersandro2000
Environment: r24106-10cc5fcd00, 23.05.5, octeon/generic, Ubiquiti EdgeRouter 4, mips64

Description:

Tailscale now provides mips64 binaries, but, the tailscale openwrt package does not have them, making installation impossible on mips64. This bug is to request the tailscale package to be updated with the mips64 binaries, to support the mips64 arch.

Additionally, the version currently in the OpenWrt repo suffers a security vulnerability (per the warning shown on the Tailscale dashboard after installation) so it should really be updated.

Thanks!

uski, 2024-10-20:
I was able to use the files in the package https://github.com/openwrt/packages/tree/master/net/tailscale and manually copying them into my mips64 system + copying the two binaries from the tailscale release, everything works fine.

So the package just needs to be updated with a newer version of tailscale and with support for mips64

mochaaP, 2024-10-20:
for the vuln part please see #24714
tldr: it should usually not affect OpenWrt use cases.

Also, we are unable to bump to a newer go release in a stable branch, but tailscale always requires the latest go. unfortunately we cannot do much about that.

this will probably land in SNAPSHOT, but not 23.05.x branches.

SuperSandro2000, 2024-10-20:
> unfortunately we cannot do much about that.

Possibilities like backporting the newer go version as a separate package is not allowed for reasons I disagree with.

SuperSandro2000, 2024-10-20:
Do we just need to drop https://github.com/openwrt/packages/blob/master/net/tailscale/Makefile#L26 for mips to work?

uski, 2024-10-20:
If we could at least make mips64 work, users should be able to install the obsolete package through opkg, then upgrade tailscale/tailscaled using its own auto-update mechanism - it's what I did on another router with a different arch.

It's very not ideal, I recall seeing quite a bit of error messages during installation, but, in the end it worked. Better than doing opkg install tailscale and being told there is no such package

SuperSandro2000, 2024-10-20:
> If we could at least make mips64 work,

It is there on snapshot https://downloads.openwrt.org/snapshots/packages/mips64_octeonplus/packages/tailscale_1.74.1-r1_mips64_octeonplus.ipk, so it is unavailable for the same reason as it is for all other platforms.

---

Please close this issue.

closed 2026-02-14 by GeorgeSapkin