openwrt/packages #25177: ocserv: log-level set much too high by default

rhester72, 2024-10-20:
Maintainer: Nikos Mavrogiannopoulos <n.mavrogiannopoulos@gmail.com>
Environment: aarch64, Raspberry Pi Compute Module 4 Rev 1.1, 23.05.5

Description: /etc/ocserv/ocserv.conf.template as included with the package sets log-level to 3, which is MUCH too noisy - suggest 0 or commenting it out for default installation

Ra2-IFV, 2024-11-13:
@nmav Can I ask what your idea is? Log level `0` is default in upstream

Ra2-IFV, 2024-11-13:
I can create a simple PR if you don't mind...

rhester72, 2024-11-13:
I don't mind...0 would work, better would probably be to comment it out in the template so it always uses the upstream default.  I can't understand the motivation of why it is set to 3 on the OpenWRT package unless the maintainer was testing something and needed debug logging and forgot to disable it after.

Ra2-IFV, 2024-11-13:
> I don't mind...0 would work, better would probably be to comment it out in the template so it always uses the upstream default.

`0` means default, should be same as being commented out, [manual page](https://ocserv.openconnect-vpn.net/ocserv.8.html)

>  I can't understand the motivation of why it is set to 3 on the OpenWRT package unless the maintainer was testing something and needed debug logging and forgot to disable it after.

Possible, hahah

nmav, 2024-11-14:
Hi, it makes sense, please submit a pull request.

Ra2-IFV, 2024-11-14:
> Hi, it makes sense, please submit a pull request.

Would it be useful to update the template conf here from the [Gitlab repo](https://gitlab.com/openconnect/ocserv/-/blob/master/doc/sample.config)? I see some differences...

Ra2-IFV, 2024-11-15:
@rhester72 you may close this issue now if you are satisfied with the change #25347

rhester72, 2024-11-15:
Closing as working, thank you!

closed 2024-11-15 by rhester72