openwrt/packages #25179: adguardhome: block and allow lists are not updated after restart

castillofrancodamian, 2024-10-21:
Whenever I reboot the device, clients receive ads even with adguardhome enabled.

Ra2-IFV, 2024-11-13:
https://github.com/openwrt/packages/blob/master/net/adguardhome/files/adguardhome.config#L3C1-L3C33

Modify `/etc/config/adguardhome` , or enable `Target Images` > `Make /var persistent` if you are using self-built image

castillofrancodamian, 2024-11-15:
@Ra2-IFV I don't think this is the solution. I don't expect it to keep logs. I expect it to get the block and allow lists after reboot. These are the configuration files.

<details><summary>cat /etc/adguardhome.yaml</summary>
<p>

```
root@netbook:~# cat /etc/adguardhome.yaml 
http:
  pprof:
    port: 6060
    enabled: false
  address: 0.0.0.0:22000
  session_ttl: 720h
users:
  - name: admin
    password: $2a$10$kjxGw2hDzwIxH5SvSq4GN.uAz2gYL5ISxD2rVeeeCyE99i9hfBRkK
auth_attempts: 5
block_auth_min: 15
http_proxy: ""
language: es
theme: dark
dns:
  bind_hosts:
    - 0.0.0.0
  port: 53
  anonymize_client_ip: false
  ratelimit: 0
  ratelimit_subnet_len_ipv4: 24
  ratelimit_subnet_len_ipv6: 56
  ratelimit_whitelist: []
  refuse_any: true
  upstream_dns:
    - tls://1.0.0.1
    - tls://1.1.1.1
    - '#tls://9.9.9.9'
    - '#tls://149.112.112.112'
    - '[/lan/]192.168.20.1'
    - '[/in-addr.arpa/]192.168.20.1'
  upstream_dns_file: ""
  bootstrap_dns:
    - 9.9.9.10
    - 149.112.112.10
    - 2620:fe::10
    - 2620:fe::fe:10
  fallback_dns: []
  upstream_mode: parallel
  fastest_timeout: 1s
  allowed_clients: []
  disallowed_clients: []
  blocked_hosts:
    - version.bind
    - id.server
    - hostname.bind
  trusted_proxies:
    - 127.0.0.0/8
    - ::1/128
  cache_size: 4194304
  cache_ttl_min: 0
  cache_ttl_max: 0
  cache_optimistic: false
  bogus_nxdomain: []
  aaaa_disabled: true
  enable_dnssec: false
  edns_client_subnet:
    custom_ip: ""
    enabled: false
    use_custom: false
  max_goroutines: 300
  handle_ddr: true
  ipset: []
[... 778 lines cut ...]
Fri Nov 15 14:10:59 2024 daemon.info avahi-daemon[2788]: New relevant interface br-lan.IPv4 for mDNS.
Fri Nov 15 14:10:59 2024 daemon.info avahi-daemon[2788]: Registering new address record for 192.168.40.2 on br-lan.IPv4.
Fri Nov 15 14:10:59 2024 daemon.notice netifd: Interface 'lan' is now up
Fri Nov 15 14:10:59 2024 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Fri Nov 15 14:10:59 2024 daemon.info avahi-daemon[2788]: Joining mDNS multicast group on interface br-lan.IPv6 with address fe80::eea8:6bff:fe52:7ee0.
Fri Nov 15 14:10:59 2024 daemon.info avahi-daemon[2788]: New relevant interface br-lan.IPv6 for mDNS.
Fri Nov 15 14:10:59 2024 daemon.info avahi-daemon[2788]: Registering new address record for fe80::eea8:6bff:fe52:7ee0 on br-lan.*.
Fri Nov 15 15:00:00 2024 daemon.err AdGuardHome[3594]: 2024/11/15 18:00:00.536277 [info] stats: warning: deleting bucket err="bucket not found"
Fri Nov 15 15:01:10 2024 authpriv.info dropbear[10037]: Child connection from 192.168.20.100:44770
Fri Nov 15 15:01:12 2024 authpriv.notice dropbear[10037]: Password auth succeeded for 'root' from 192.168.20.100:44770
Fri Nov 15 15:01:30 2024 daemon.err uhttpd[2686]: [info] luci: accepted login on / for root from 192.168.20.100
Fri Nov 15 15:06:23 2024 daemon.err AdGuardHome[3594]: 2024/11/15 18:06:23.343597 [info] auth: user "admin" successfully logged in from ip 192.168.20.100
Fri Nov 15 15:06:24 2024 daemon.err AdGuardHome[3594]: 2024/11/15 18:06:24.777525 [info] home: language is set to es
Fri Nov 15 15:06:24 2024 daemon.err AdGuardHome[3594]: 2024/11/15 18:06:24.777675 [info] home: theme is set to dark
Fri Nov 15 15:06:36 2024 authpriv.info dropbear[10037]: Exit (root) from <192.168.20.100:44770>: Disconnect received
Fri Nov 15 15:06:38 2024 authpriv.info dropbear[10726]: Child connection from 192.168.20.100:44006
Fri Nov 15 15:06:40 2024 authpriv.notice dropbear[10726]: Password auth succeeded for 'root' from 192.168.20.100:44006
´´´

</p>
</details> 

Could this be the reason why the lists are not downloaded?

![Captura de pantalla_20241115_150923](https://github.com/user-attachments/assets/f588817d-c04a-4272-9970-6efaecb92d92)

Ra2-IFV, 2024-11-16:
The working directory is the place where AdGuardHome stores everything in, including databases and filters, if you didn't modify default settings. If you don't understand how this trick works this is because `/var` is a symbolic link to `/tmp` in OpenWrt, and `/tmp` saves everything in the DRAM. 
Priority shouldn't be related to this issue.

Ra2-IFV, 2024-11-16:
The working directory is the place where AdGuardHome saves everything, including query logs and filyers, if you are not modifying default settings.
You are seeing failures about downloading filters because OpenWrt has not connected to the network, should not be related to this.

Ra2-IFV, 2024-11-16:
The working directory is the place where AdGuardHome saves everything, including query logs and filyers, if you are not modifying default settings.
You are seeing failures about downloading filters because OpenWrt has not connected to the network, should not be related to this.

On November 15, 2024 6:11:08 PM UTC, Franco Castillo ***@***.***> wrote:
>@Ra2-IFV I don't think this is the solution. I don't expect it to keep logs. I expect it to get the block and allow lists after reboot. These are the configuration files.
>
><details><summary>cat /etc/adguardhome.yaml</summary>
><p>
>
>```
>***@***.***:~# cat /etc/adguardhome.yaml 
>http:
>  pprof:
>    port: 6060
>    enabled: false
>  address: 0.0.0.0:22000
>  session_ttl: 720h
>users:
>  - name: admin
>    password: $2a$10$kjxGw2hDzwIxH5SvSq4GN.uAz2gYL5ISxD2rVeeeCyE99i9hfBRkK
>auth_attempts: 5
>block_auth_min: 15
>http_proxy: ""
>language: es
>theme: dark
>dns:
>  bind_hosts:
>    - 0.0.0.0
>  port: 53
>  anonymize_client_ip: false
>  ratelimit: 0
>  ratelimit_subnet_len_ipv4: 24
>  ratelimit_subnet_len_ipv6: 56
>  ratelimit_whitelist: []
>  refuse_any: true
>  upstream_dns:
>    - tls://1.0.0.1
>    - tls://1.1.1.1
>    - '#tls://9.9.9.9'
>    - '#tls://149.112.112.112'
>    - '[/lan/]192.168.20.1'
>    - '[/in-addr.arpa/]192.168.20.1'
>  upstream_dns_file: ""
>  bootstrap_dns:
>    - 9.9.9.10
>    - 149.112.112.10
>    - 2620:fe::10
>    - 2620:fe::fe:10
>  fallback_dns: []
>  upstream_mode: parallel
>  fastest_timeout: 1s
>  allowed_clients: []
>  disallowed_clients: []
>  blocked_hosts:
>    - version.bind
>    - id.server
>    - hostname.bind
>  trusted_proxies:
>    - 127.0.0.0/8
>    - ::1/128
>  cache_size: 4194304
>  cache_ttl_min: 0
>  cache_ttl_max: 0
>  cache_optimistic: false
>  bogus_nxdomain: []
>  aaaa_disabled: true
>  enable_dnssec: false
>  edns_client_subnet:
>    custom_ip: ""
>    enabled: false
[... 693 lines cut ...]
>Fri Nov 15 14:10:42 2024 daemon.info procd: Not starting instance acpid::data, command not set
>Fri Nov 15 14:10:42 2024 daemon.info procd: - init complete -
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7182] N: ttyd 1.7.3 (libwebsockets 4.3.3-unknown)
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7183] N: tty configuration:
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7184] N:   start command: /bin/login
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7185] N:   close signal: SIGHUP (1)
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7185] N:   terminal type: xterm-256color
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7359] N:    /usr/lib/libwebsockets-evlib_uv.so
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7361] N: lws_create_context: LWS: 4.3.3-unknown, NET CLI SRV H1 H2 WS ConMon IPV6-off
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7363] N: elops_init_pt_uv:  Using foreign event loop...
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7365] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7366] N: __lws_lc_tag:  ++ [vh|0|netlink] (1)
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7368] N: __lws_lc_tag:  ++ [vh|1|default||7681] (2)
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7370] N: [vh|1|default||7681]: lws_socket_bind: source ads 0.0.0.0
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7371] N: __lws_lc_tag:  ++ [wsi|1|listen|default||7681] (2)
>Fri Nov 15 14:10:42 2024 daemon.notice ttyd[3482]: [2024/11/15 14:10:42:7372] N:  Listening on port: 7681
>Fri Nov 15 14:10:47 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:47.729239 [info] AdGuard Home, version v0.107.53
>Fri Nov 15 14:10:47 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:47.855985 [info] tls: using default ciphers
>Fri Nov 15 14:10:47 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:47.862771 [info] safesearch default: disabled
>Fri Nov 15 14:10:48 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:48.148586 [info] Initializing auth module: /tmp/adguardhome/data/sessions.db
>Fri Nov 15 14:10:48 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:48.149531 [info] auth: initialized.  users:1  sessions:0
>Fri Nov 15 14:10:48 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:48.149633 [info] AdGuard Home updates are disabled
>Fri Nov 15 14:10:48 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:48.149670 [info] web: initializing
>Fri Nov 15 14:10:48 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:48.170104 [info] dnsproxy: upstream mode is set mode=parallel
>Fri Nov 15 14:10:48 2024 daemon.err AdGuardHome[3594]: 2024/11/15 17:10:48.170274 [info] dnsproxy: cache enabled siz

castillofrancodamian, 2024-11-26:
@Ra2-IFV Is this the information you needed to know?

```
root@netbook:~# cat /etc/config/adguardhome 
config adguardhome config
        # Where to store persistent data by AdGuard Home
        option workdir /var/adguardhome
```

I think this is the default setting as I don't remember modifying that file.

Ra2-IFV, 2024-11-27:
yeah just change `/var/adguardhome` to somewhere else on your storage device, the comment should be clear enough

castillofrancodamian, 2024-11-27:
@Ra2-IFV But wouldn't making that change do the opposite of what one would expect? That is, prevent writing to flash memory all the time?

Ra2-IFV, 2024-11-27:
Just change it and then AdGuardHome will save datas in the disk (in your case it's SD card, eMMC or somewhat), just try it...

castillofrancodamian, 2024-11-27:
@Ra2-IFV But that is not what is desired. What is desired is that the statistics are not saved after rebooting the device since it would drain the micro SD. And I just created a clean instance of OpenWrt in a virtual machine and I see that the lists are updated automatically after rebooting it.

<details><summary>cat /etc/adguardhome.yaml</summary>
<p>

```
root@OpenWrt:~# cat /etc/adguardhome.yaml 
http:
  pprof:
    port: 6060
    enabled: false
  address: 0.0.0.0:8080
  session_ttl: 720h
users:
  - name: root
    password: $2a$10$RQ0iTwcge8vV/3glMhaGc.EbCPbjlrSPBpvMskJB8OOFG.R4C2Ine
auth_attempts: 5
block_auth_min: 15
http_proxy: ""
language: ""
theme: auto
dns:
  bind_hosts:
    - 0.0.0.0
  port: 5300
  anonymize_client_ip: false
  ratelimit: 20
  ratelimit_subnet_len_ipv4: 24
  ratelimit_subnet_len_ipv6: 56
  ratelimit_whitelist: []
  refuse_any: true
  upstream_dns:
    - 1.0.0.1
    - 1.1.1.1
  upstream_dns_file: ""
  bootstrap_dns:
    - 9.9.9.10
    - 149.112.112.10
    - 2620:fe::10
    - 2620:fe::fe:10
  fallback_dns: []
  upstream_mode: load_balance
  fastest_timeout: 1s
  allowed_clients: []
  disallowed_clients: []
  blocked_hosts:
    - version.bind
    - id.server
    - hostname.bind
  trusted_proxies:
    - 127.0.0.0/8
    - ::1/128
  cache_size: 4194304
  cache_ttl_min: 0
  cache_ttl_max: 0
  cache_optimistic: false
  bogus_nxdomain: []
  aaaa_disabled: false
  enable_dnssec: false
  edns_client_subnet:
    custom_ip: ""
    enabled: false
    use_custom: false
  max_goroutines: 300
  handle_ddr: true
  ipset: []
  ipset_file: ""
  bootstrap_prefer_ipv6: false
  upstream_timeout: 10s
  private_networks: []
  use_private_ptr_resolvers: true
  local_ptr_upstreams: []
  use_dns64: false
  dns64_prefixes: []
  serve_http3: false
  use_http3_upstreams: false
  serve_plain_dns: true
  hostsfile_enabled: true
tls:
  enabled: false
  server_name: ""
  force_https: false
  port_https: 443
  port_dns_over_tls: 853
  port_dns_over_quic: 853
  port_dnscrypt: 0
  dnscrypt_config_file: ""
  allow_unencrypted_doh: false
  certificate_chain: ""
  private_key: ""
  certificate_path: ""
  private_key_path: ""
  strict_sni_check: false
querylog:
  dir_path: ""
  ignored: []
  interval: 2160h
  size_memory: 1000
  enabled: true
  file_enabled: true
statistics:
  dir_path: ""
  ignored: []
  interval: 2160h
  enabled: true
filters:
  - enabled: true
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_1.txt
    name: AdGuard DNS filter
    id: 1
  - enabled: true
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_2.txt
    name: AdAway Default Blocklist
    id: 2
  - enabled: true
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_27.txt
    name: OISD Blocklist Big
    id: 1732736316
  - enabled: true
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_23.txt
    name: WindowsSpyBlocker - Hosts spy rules
    id: 1732736317
  - enabled: true
    url: https://adguardteam.github.io/HostlistsRegistry/assets/filter_7.txt
    name: Perflyst and Dandelion Sprout's Smart-TV Blocklist
    id: 1732736319
whitelist_filters:
  - enabled: true
    url: https://raw.githubusercontent.com/anudeepND/whitelist/master/domains/whitelist.txt
    name: permitidos
    id: 1732736323
user_rules: []
dhcp:
  enabled: false
  interface_name: ""
  local_domain_name: lan
  dhcpv4:
    gateway_ip: ""
    subnet_mask: ""
    range_start: ""
    range_end: ""
    lease_duration: 86400
    icmp_timeout_msec: 1000
    options: []
  dhcpv6:
    range_start: ""
    lease_duration: 86400
    ra_slaac_only: false
    ra_allow_slaac: false
filtering:
  blocking_ipv4: ""
  blocking_ipv6: ""
  blocked_services:
    schedule:
      time_zone: UTC
    ids: []
  protection_disabled_until: null
  safe_search:
    enabled: false
    bing: true
    duckduckgo: true
    ecosia: true
    google: true
    pixabay: true
    yandex: true
    youtube: true
  blocking_mode: default
  parental_block_host: family-block.dns.adguard.com
  safebrowsing_block_host: standard-block.dns.adguard.com
  rewrites: []
  safe_fs_patterns:
    - /tmp/adguardhome/userfilters/*
  safebrowsing_cache_size: 1048576
  safesearch_cache_size: 1048576
  parental_cache_size: 1048576
  cache_time: 30
  filters_update_interval: 24
  blocked_response_ttl: 10
  filtering_enabled: true
  parental_enabled: false
  safebrowsing_enabled: false
  protection_enabled: true
clients:
  runtime_sources:
    whois: true
    arp: true
    rdns: true
    dhcp: true
    hosts: true
  persistent: []
log:
  enabled: true
  file: ""
  max_backups: 0
  max_size: 100
  max_age: 3
  compress: false
  local_time: false
  verbose: false
os:
  group: ""
  user: ""
  rlimit_nofile: 0
schema_version: 29
```

</p>
</details> 

```
root@OpenWrt:~# cat /etc/config/adguardhome 
config adguardhome config
        # Where to store persistent data by AdGuard Home
        option workdir /var/adguardhome
        option config /etc/adguardhome.yaml
```

Ra2-IFV, 2024-11-28:
So you wish to keep block lists while writing statistic database to memory?
Looks AdGuardHome itself has this line:

Ra2-IFV, 2024-11-28:
You can also use symbolic links to let program write to the place you want, it's up to you.
I think you can close this issue for now.

castillofrancodamian, 2025-02-24:
It should be fixed now. When I restart the device, I see that the lists are updated.

closed 2025-02-24 by castillofrancodamian