openwrt/packages #25194: protobuf: Update to the latest version

tiagogaspar8, 2024-10-24:
Maintainer: Ken Keys <kkeys@caida.org> @kenkeys 
Environment: All

Description:

The protobuf package is extremely out of date. I have tried to update it but unfortunately, I don't have enough knowledge to do it myself, and as I'm trying to update the netdata package, it needs the latest version.

tiagogaspar8, 2024-11-13:
I managed to compile protobuf, I will submit a pull request as soon as I can

tiagogaspar8, 2024-11-13:
You don't need that, protobuff automatically pulls, compiles and installs abseil, I can share the makefile in a pull request later today, I'd appreciate your input 😁

vortexilation, 2024-11-13:
How about separating abseil-cpp dependency from protobuf ? , this will make version pinning and updating easier without dependent to each other : 
```
# SPDX-License-Identifier: GPL-2.0-only
#
# Copyright (C) 2021 Martin Schneider <martschneider@google.com>

include $(TOPDIR)/rules.mk

PKG_NAME:=abseil-cpp
PKG_VERSION:=20240116.1
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/abseil/abseil-cpp.git
PKG_MIRROR_HASH:=1afa569daf7268a724924ab98b54af8cdbbefa33dc8e044ce5df9bd5e708f640
PKG_SOURCE_DATE:=2024-11-08
PKG_SOURCE_VERSION:=2f9e432cce407ce0ae50676696666f33a77d42ac

PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/cmake.mk

define Package/abseil-cpp
	SECTION:=libs
	CATEGORY:=Libraries
	TITLE:=Abseil Common Libraries
	URL:=https://github.com/abseil/abseil-cpp
	DEPENDS:=+libstdcpp
endef

define Package/abseil-cpp/description
	The repository contains the Abseil C++ library code. 
	Abseil is an open-source collection of C++ code (compliant to C++14) designed to augment the C++ standard library.
endef

CMAKE_OPTIONS+=-DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON

define Package/abseil-cpp/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/* $(1)/usr/lib/
endef

define Build/InstallDev
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/* $(1)/usr/lib
	$(INSTALL_DIR) $(1)/usr/include
	$(CP) $(PKG_INSTALL_DIR)/usr/include/* $(1)/usr/include
endef

$(eval $(call BuildPackage,abseil-cpp))
```

```
diff -Naur a/libs/protobuf/Makefile b/libs/protobuf/Makefile
--- a/libs/protobuf/Makefile
+++ b/libs/protobuf/Makefile
@@ -8,19 +8,21 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=protobuf
-PKG_VERSION:=3.17.3
-PKG_RELEASE:=2
+PKG_VERSION:=25.3
+PKG_RELEASE:=1
 
-PKG_SOURCE:=$(PKG_NAME)-cpp-$(PKG_VERSION).tar.gz
-PKG_SOURCE_URL:=https://github.com/google/protobuf/releases/download/v$(PKG_VERSION)
-PKG_HASH:=51cec99f108b83422b7af1170afd7aeb2dd77d2bcbb7b6bad1f92509e9ccf8cb
[... 147 lines cut ...]
	-DBUILD_PROTOC=ON \
	-DCMAKE_SKIP_RPATH=OFF \
	-DCMAKE_INSTALL_RPATH="${STAGING_DIR}/usr/lib"

define Build/InstallDev
	$(INSTALL_DIR) $(1)/usr/include
	$(CP) $(PKG_INSTALL_DIR)/usr/include/* $(1)/usr/include
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/* $(1)/usr/lib/
	$(INSTALL_DIR) $(1)/usr/bin
	$(CP) $(PKG_INSTALL_DIR)/usr/bin/protoc-gen-c $(1)/usr/bin
	$(LN) protoc-gen-c $(1)/usr/bin/protoc-c
endef

define Package/libprotobuf-c/install
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/libprotobuf-c.so* $(1)/usr/lib/
	
endef

$(eval $(call BuildPackage,libprotobuf-c))
$(eval $(call HostBuild))
```

I suggest to use protobuf version 25.3 or 26.0 as version above [it is not supported by protobuf-c](https://github.com/protobuf-c/protobuf-c/tree/edmonds/google-protobuf-26-fixes)

tiagogaspar8, 2024-11-13:
You managed to compile netdata 2.0? 😲 
I have a pull request open here for aged and I was looking for help on how to get it to build...
https://github.com/openwrt/packages/pull/22677

Ra2-IFV, 2024-11-13:
I fixed the patch for protobuf 28.3, if anyone needs it

> I suggest to use protobuf version 25.3 or 26.0 as version above [it is not supported by protobuf-c](https://github.com/protobuf-c/protobuf-c/tree/edmonds/google-protobuf-26-fixes)

Might need refreshing if you are updating to a different version  
```diff
--- a/cmake/install.cmake
+++ b/cmake/install.cmake
@@ -35,8 +35,8 @@ endif ()
 
 foreach(_library ${_protobuf_libraries})
   if (UNIX AND NOT APPLE)
-    set_property(TARGET ${_library}
-      PROPERTY INSTALL_RPATH "$ORIGIN")
+    # set_property(TARGET ${_library}
+    #   PROPERTY INSTALL_RPATH "$ORIGIN")
   elseif (APPLE)
     set_property(TARGET ${_library}
       PROPERTY INSTALL_RPATH "@loader_path")
@@ -62,8 +62,8 @@ if (protobuf_BUILD_PROTOC_BINARIES)
   endif ()
   foreach (binary IN LISTS _protobuf_binaries)
     if (UNIX AND NOT APPLE)
-      set_property(TARGET ${binary}
-        PROPERTY INSTALL_RPATH "$ORIGIN/../${CMAKE_INSTALL_LIBDIR}")
+      # set_property(TARGET ${binary}
+      #   PROPERTY INSTALL_RPATH "$ORIGIN/../${CMAKE_INSTALL_LIBDIR}")
     elseif (APPLE)
       set_property(TARGET ${binary}
         PROPERTY INSTALL_RPATH "@loader_path/../lib")
```

vortexilation, 2024-11-13:
@tiagogaspar8 
Yes, but forgot to mention it's all on glibc, not musl, i will try musl now.
Probably harder on musl.

About your post on that PR, i am not encountering the same thing on glibc.

You could open up protobuf PR based on my finding if you want.

Here is the rough draft for netdata 2.0 package and libs package : 
[netdata.zip](https://github.com/user-attachments/files/17736948/netdata.zip)

The netdata 2.0 package from me is different, it using cmake build system instead.

vortexilation, 2024-11-13:
@tiagogaspar8 
Your previous issue on netdata repo, if I am not mistaken I have encountered it before, try to not using bundled abseil-cpp/protobuf from netdata.
Use it's own external abseil-cpp and protobuf OpenWrt package like above.

tiagogaspar8, 2024-11-23:
Hey @vortexilation,
I was trying your suggestion but it's not working, and now I noticed you mentioned it's working for you when targeting x86/64, could you try your setup targeting another arch besides x86?
Thanks!

vortexilation, 2024-11-23:
@tiagogaspar8 
Trying to build for aarch64_cortex-a53_musl now, but my build environment quite messy as currently testing for kernel 6.12.

Could you try to locate "ld-musl-aarch64.so.1" in your staging_dir directory?. I assume it's on $(STAGING_DIR_ROOT)/lib ?.

vortexilation, 2024-11-23:
```
/mnt/openwrt/ss_mod/staging_dir/target-aarch64_cortex-a53_musl/usr/bin/protoc-28.3.0 -I/mnt/openwrt/ss_mod/build_dir/target-aarch64_cortex-a53_musl/netdata-v2.0.0/src/aclk/aclk-schemas --cpp_out=/mnt/openwrt/ss_mod/build_dir/target-aarch64_cortex-a53_musl/netdata-v2.0.0/src/aclk/aclk-schemas /mnt/openwrt/ss_mod/build_dir/target-aarch64_cortex-a53_musl/netdata-v2.0.0/src/aclk/aclk-schemas/proto/aclk/v1/lib.proto
```
I think it failed to run protoc successfully because of could not found ld-musl-aarch64.so.1 dependency.

tiagogaspar8, 2024-11-23:
BTW, forgot to say I found the mentioned lib in `staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/lib`

> I think it failed to run protoc successfully because of could not found ld-musl-aarch64.so.1 dependency.

Yes, I just thought it was looking for it in the wrong place because the error states it's looking for it in the /lib, which for me means root filesystem of the building host, am I wrong?

> Where do you put the line?

I put it above the CMAKE_OPTIONS as in your makefile

vortexilation, 2024-11-23:
> BTW, forgot to say I found the mentioned lib in `staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/lib`

Yes that's exactly what $(STAGING_DIR_ROOT)/lib is.

> Yes, I just thought it was looking for it in the wrong place because the error states it's looking for it in the /lib, which for me means root filesystem of the building host, am I wrong?

Yes, you are correct, the netdata build system not also looking at the OpenWrt toolchain, $(STAGING_DIR_ROOT)/lib , which is ld-musl-aarch64.so.1 located.

Btw what is your target device?

tiagogaspar8, 2024-11-23:
The device I'm currently building for is Gl.inet MT6000, Mediatek Filogic.
Did your build succeed?

vortexilation, 2024-11-23:
Yes, my build is succeed (NXP i.MX8M), even with latest netdata 2.0.3.
Probably the issue is with your build host/os ?, I don't encounter the ld-musl-aarch64.so.1 not found error.
I am using debian sid.

Changes : 
Protobuf Makefile
```
#
# Copyright (C) 2007-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=protobuf
PKG_VERSION:=25.3
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/protocolbuffers/protobuf.git
PKG_MIRROR_HASH:=63ddc2978bc7cdd357f66ed70f273a787960e990a328887b7f65d119094c7353
PKG_SOURCE_DATE:=2024-11-08
PKG_SOURCE_VERSION:=4a2aef570deb2bfb8927426558701e8bfc26f2a4

PKG_MAINTAINER:=Ken Keys <kkeys@caida.org>
PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=LICENSE
PKG_CPE_ID:=cpe:/a:google:protobuf

CMAKE_BINARY_SUBDIR:=build

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/host-build.mk
include $(INCLUDE_DIR)/cmake.mk

define Package/protobuf/Default
  SECTION:=libs
  CATEGORY:=Libraries
  TITLE:=A structured data encoding library
  URL:=https://github.com/google/protobuf
  DEPENDS:=+zlib +libpthread +libatomic +libstdcpp +abseil-cpp
endef

define Package/protobuf
  $(call Package/protobuf/Default)
  DEPENDS+=+protobuf-lite
endef

define Package/protobuf-lite
  $(call Package/protobuf/Default)
endef

define Package/protobuf/description/Default
Protocol Buffers are a way of encoding structured data in an efficient
yet extensible format. Google uses Protocol Buffers for almost all
of its internal RPC protocols and file formats.
endef

define Package/protobuf/description
$(call Package/protobuf/description/Default)

This package provides the libprotoc, libprotobuf, and libprotobuf-lite
libraries.  For a much smaller protobuf package, see "protobuf-lite".

endef

define Package/protobuf-lite/description
$(call Package/protobuf/description/Default)
[... 144 lines cut ...]
	touch $(1)/etc/netdata/.opt-out-from-anonymous-statistics
	$(INSTALL_DIR) $(1)/usr/lib
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/netdata $(1)/usr/lib
	$(CP) $(1)/usr/lib/netdata/conf.d/health_alarm_notify.conf $(1)/etc
	$(INSTALL_DIR) $(1)/usr/libexec/netdata/plugins.d
	$(CP) $(PKG_INSTALL_DIR)/usr/libexec/netdata/plugins.d/tc-qos-helper.sh $(1)/etc
	$(INSTALL_DIR) $(1)/usr/sbin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/sbin/netdata $(1)/usr/sbin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/sbin/netdatacli $(1)/usr/sbin
	$(INSTALL_DIR) $(1)/usr/share/netdata
	$(CP) $(PKG_INSTALL_DIR)/usr/share/netdata $(1)/usr/share
	rm $(1)/usr/share/netdata/web/demo*html
	rm $(1)/usr/share/netdata/web/fonts/*.svg
	rm $(1)/usr/share/netdata/web/fonts/*.ttf
	rm $(1)/usr/share/netdata/web/fonts/*.woff
	rm $(1)/usr/share/netdata/web/images/*.png
	rm $(1)/usr/share/netdata/web/images/*.gif
	rm $(1)/usr/share/netdata/web/images/*.ico
	rm -rf $(1)/usr/share/netdata/web/old
	$(INSTALL_DIR) $(1)/etc/init.d
	$(INSTALL_BIN) ./files/netdata.init $(1)/etc/init.d/netdata
endef

$(eval $(call BuildPackage,netdata))
```

vortexilation, 2024-11-23:
My solution with export LD_LIBRARY_PATH=$(STAGING_DIR_ROOT)/lib is wrong for aarch64 cross-compiling, it should be fixed protoc binary which build for host (x86_64) not current protoc binary which build for aarch64, my latest changes above should fixes it up.

protobuf for host need to be re-build with : 
```
make package/protobuf/host/{clean,compile} -j9 V=s 2>&1 | tee build.log
```
Then continue with netdata re-build.

tiagogaspar8, 2024-11-24:
Good news! after some testing, I managed to compile!
I am now testing running it in my router, and I've faced a new bug when claiming with the token, have you faced this?
`msg="CLAIM: Cannot generate RSA key, EVP_PKEY_keygen() failed"`
Regarding Protobud, do you have any reason to have abseil as a separate package? Using your makefile I was not able to build protobuf + abseil, so If I'm going to submit a patch to update protobud it will probably be with the only makefile I can compile 😕

vortexilation, 2024-11-24:
Glad that you can build it.

> I am now testing running it in my router, and I've faced a new bug when claiming with the token, have you faced this?
> `msg="CLAIM: Cannot generate RSA key, EVP_PKEY_keygen() failed"`

This is happening when running the netdata?

Regarding separate abseil-cpp, If i am not mistaken it was about a package failed to compile if using bundled abseil-cpp, but I cannot recall exactly which package is it.

About protobuf, I suggest to keep it with version 25.3, as above 26.0 will make libprotobuf-c [problematic ](https://github.com/protobuf-c/protobuf-c/pull/711)when compiling it.
Also netdata only [needs max protobuf 25.3](https://github.com/netdata/netdata/blob/c8d4cc2a2e8640c59cf47ddb05932360ad300f78/packaging/cmake/Modules/NetdataProtobuf.cmake#L15) for compilation.

tiagogaspar8, 2024-11-25:
> This is happening when running the netdata?

Yes, when installed on my router and trying to claim the agent in netdata cloud with the token.

> Regarding separate abseil-cpp, If i am not mistaken it was about a package failed to compile if using bundled abseil-cpp, but I cannot recall exactly which package is it.

If you remember let me know so I can find it and troubleshoot it

> About protobuf, I suggest to keep it with version 25.3, as above 26.0 will make libprotobuf-c https://github.com/protobuf-c/protobuf-c/pull/711when compiling it.
> Also netdata only [needs max protobuf 25.3](https://github.com/netdata/netdata/blob/c8d4cc2a2e8640c59cf47ddb05932360ad300f78/packaging/cmake/Modules/NetdataProtobuf.cmake#L15) for compilation.

I'll keep it in version 25 for now, but I'll push it to 25.5, which is the latest bugfix release.
I'm not sure how protobuf-c can be kept, it hasn't been updated in almost a year.

vortexilation, 2024-11-25:
> Yes, when installed on my router and trying to claim the agent in netdata cloud with the token.

Haven't tested netdata thoroughly , maybe  you could report it upstream.

There is quite a lot of package [depending ](https://github.com/search?q=repo%3Aopenwrt%2Fpackages%20libprotobuf-c&type=code)on protobuf-c , for protobuf 25.3 supports in protobuf-c you need [this ](https://github.com/protobuf-c/protobuf-c/tree/edmonds/google-protobuf-26-fixes)specific branch.

vortexilation, 2024-12-06:
@tiagogaspar8 

Node 22.11.0 seems has [issue ](https://github.com/openwrt/packages/issues/25483)with bundled abseil-cpp from protobuf 25.3.

tiagogaspar8, 2024-12-25:
Hi @vortexilation,
I finally had some minutes to look at this again, it seems the protobuf proposed to be merged doesn't seem to build, do you have a version that builds and works? The one you posted here some time ago doesn't work anymore.
Thanks.

tiagogaspar8, 2025-05-15:
Hi all,

@aleks-mariusz, actually, this has been a tough endeavour as the netdata team is anything but useful, insisting on compiling the software and libraries locally upon each install and not giving any documentation for distro maintainers, everything behind a iffy install script.
I have been successfull with the awesome help of @vortexilation in pouting the software up and running, working on gettin the features packaged up.

Unfortunately, with the latest version 2.5.1 they have added yet another dependency that makes the code failing to compile... not sure if @vortexilation has had any luck with that version.