openwrt/packages #25204: docker-compose: The package is missing for OpenWrt 23.05

eduhoribe, 2024-10-25:
Maintainer: @jmarcet
Environment: aarch64_generic, NanoPi R2S, 23.05.0

Unable to install the `docker-compose` package on OpenWrt 23.05.0.
Ran `opkg update` and then `opkg install docker-compose` which returned:
```sh
Unknown package 'docker-compose'.
Collected errors:
 * opkg_install_cmd: Cannot install package docker-compose.
```

Judging by [this](https://downloads.openwrt.org/releases/faillogs-23.05/aarch64_generic/packages/docker-compose/compile.txt), it seems to be a build error:
```
...
Finding targets
verifying github.com/aws/aws-sdk-go-v2/config@v1.15.5/go.mod: checksum mismatch
	downloaded: h1:G7mAYYxgmS0lVkHyy2hEOLQCFB0DlQFTMLWggykrydY=
	go.sum:     h1:ZijHHh0xd/A+ZY53az0qzC5tT46kt4JVCePf2NX9Lk4=

SECURITY ERROR
This download does NOT match an earlier download recorded in go.sum.
The bits may have been replaced on the origin server, or an attacker may
have intercepted the download attempt.

For more information, see 'go help module-auth'.

Building targets
no Go files in /builder/shared-workdir/build/sdk/build_dir/target-aarch64_generic_musl/compose-2.18.1/.go_work/build

make[3]: *** [Makefile:50: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_generic_musl/compose-2.18.1/.built] Error 1
time: package/feeds/packages/docker-compose/compile#1.66#2.50#6.86
```

jmarcet, 2024-10-27:
Hi @eduhoribe,

I'm not sure about that error but the package is available both for [23.05.0](https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_generic/packages/docker-compose_2.18.1-1_aarch64_generic.ipk) and the newest stable [23.05.5](https://downloads.openwrt.org/releases/23.05.5/packages/aarch64_generic/packages/docker-compose_2.18.1-1_aarch64_generic.ipk)

drewlloyd, 2024-10-30:
Same problem - security error - but for [aarch64_cortex-a53](https://downloads.openwrt.org/releases/faillogs-23.05/aarch64_cortex-a53/packages/docker-compose/compile.txt)

Gr4ffy, 2024-11-04:
Same here for x86_64, package is missing.
https://downloads.openwrt.org/releases/faillogs-23.05/x86_64/packages/docker-compose/compile.txt

1715173329, 2024-11-22:
Network issue, now builds fine on buildbot.

closed 2024-11-22 by 1715173329