openwrt/packages #25207: pptpd: build error: implicit declaration of function 'logwtmp'

M95D, 2024-10-26:
Maintainer: none, last update by @ptpt52 last week, commited by @hnyman
Environment: armv7, mvebu Armada 385, Turris Omnia, OpenWrt git main

Description:
```
make[5]: Entering directory '/var/tmp/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/pptpd-1.5.0/plugins'
ccache arm-openwrt-linux-muslgnueabi-gcc -o pptpd-logwtmp.so -shared -pipe -fno-caller-saves -fno-plt -pipe -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/var/tmp/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/pptpd-1.5.0=pptpd-1.5.0 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector-all -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I.. -I../../include -fPIC pptpd-logwtmp.c -lutil
pptpd-logwtmp.c: In function 'ip_up':
pptpd-logwtmp.c:54:3: error: implicit declaration of function 'logwtmp' [-Wimplicit-function-declaration]
   54 |   logwtmp(ifname, user, pptpd_original_ip);
      |   ^~~~~~~
make[5]: *** [Makefile:19: pptpd-logwtmp.so] Error 1
```

M95D, 2024-10-26:
PS:
I've accepted the ideea that nobody does manual tests anymore, but I'm really interested to know why this wasn't caught by the automated tests.

hnyman, 2024-10-26:
> I'm really interested to know why this wasn't detected by the automated tests.

Maybe because buildbot builds it quite happily, also for the target that you are using.
It is not in the faillogs...
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/

...and can be found from the download directory, built two days ago for the last time:
pptpd_1.5.0-r1_arm_cortex-a9_vfpv3-d16.ipk | 31.0 KB | Thu Oct 24 23:38:33 2024
https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/pptpd_1.5.0-r1_arm_cortex-a9_vfpv3-d16.ipk

M95D, 2024-10-26:
Yes, that's what I'm talking about.
I can reproduce the error with a fresh clone of git repo, with only my .config added. Why is it that automated builds *don't* fail?
[config.txt](https://github.com/user-attachments/files/17531676/config.txt)

brada4, 2024-10-27:
Run diffconfig.sh please

neheb, 2024-10-27:
logwtmp being implicit becomes an error with GCC14.

closed 2024-10-30 by neheb