openwrt/packages #25209: qemu-ga: segmentation fault on VM shutdown from host

val-kulkov, 2024-10-27:
Maintainer: @yousong 
Environment: x86_64, generic, OpenWrt SNAPSHOT r27910-1dc86af356
Package version:
qemu-firmware-seabios - 9.1.0-r2
qemu-ga - 9.1.0-r2

Description:
A host-initiated shutdown of a VM with OpenWrt never completes. The kvm process running on the host has to be manually killed for the VM to stop.

qemu-ga running in the foreground on the guest OpenWrt machine produces the following on stdout:

```
...
1730056540.243011: debug: received EOF
1730056540.343262: debug: read data, count: 109, data: {"arguments":{"id":3042929038},"execute":"guest-sync-delimited"}
{"arguments":{},"execute":"guest-shutdown"}

1730056540.343402: debug: process_event: called
1730056540.343427: debug: processing command
1730056540.343492: debug: sending data, count: 24
1730056540.343639: debug: process_event: called
1730056540.343680: debug: processing command
1730056540.343773: debug: g_unix_open_pipe() called with FD_CLOEXEC; please migrate to using O_CLOEXEC instead
Segmentation fault
```

I believe the second last line in the output above gives a hint about what might be going wrong.

janh, 2025-01-17:
I noticed the same issue on OpenWrt 24.10.0-rc5.

A message like this appears on the console, and the VM does not shut down:
```
[   90.391011] traps: qemu-ga[3585] general protection fault ip:5631794ca3e7 sp:7fffcd4a7070 error:0 in qemu-ga[5631794b5000+1d000]
```

This seems to be an issue with shutdown patch which was recently updated by @vooon (https://github.com/openwrt/packages/pull/25053, https://github.com/openwrt/packages/pull/25106).

The current version of the patch from Alpine works fine:
https://gitlab.alpinelinux.org/alpine/aports/-/blob/b720d51ec844d4754dd5b29084350aa1f5c9a74d/community/qemu/guest-agent-shutdown.patch

vooon, 2025-01-17:
Hmm, i think the only possible cause - fallback_cmd remain nil.
I may update the patch, anyway we always fallback on the wrt.

janh, 2025-01-18:
@vooon: I debugged this using valgrind. It is actually caused by another bug in QEMU (https://github.com/qemu/qemu/commit/9cfe110d9fc0be88178770a85dc6170eecdf6be1). The fix is also included in QEMU 9.1.2, so I think it would make sense to cherry-pick https://github.com/openwrt/packages/commit/6ee7a479bc3a7414e299e7f51893c3de8e54aa3a to the 24.10 branch.

In addition, I also found an unrelated bug in the shutdown patch: `local_err` is not cleared in the fallback branch, so while shutdown actually does work if the patch for `ga_pipe_read_str` is applied, the error from the first call `ga_run_command` is still returned (so the host thinks the shutdown failed when it actually worked). This could be fixed by calling `error_free_or_abort(&local_err);`.

But I see you already created a pull request to switch to the patch from Alpine. That of course fixes the second bug as well.

vooon, 2025-01-18:
@janh aha, thanks. Regarding backporting - sorry, i'm always on the master, so never ever done that for wrt :)
So i suggest you may do that.

janh, 2025-01-18:
@vooon No problem, I'll create a pull request for that once #25780 has been applied.

commit 1268618b (2025-04-24): qemu: fix guest agent patch

Replace to fix #25209
commit 3f3730c0 (2025-04-28): qemu: fix guest agent patch

Replace to fix #25209

(cherry picked from commit 1268618b49e9fdbea380d625a946c364a90eff41)