openwrt/packages #25210: tinyproxy does not start on boot

mal19992, 2024-10-27:
I use tinyproxy - 1.11.1-r3 on OpenWrt SNAPSHOT, r27913-a2aabc9a7a

tinyproxy fails to start on boot. However, it can be started OK manually.
`service tinyproxy restart
`
I managed to start it from rc.local with 
```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

(
set -x
echo "START rc.local"
sleep 60
echo "STARTED rc.local"
service tinyproxy restart
echo "STARTED rc.local DONE"
) &

exit 0
```
If I remove sleep 60 -- then tinyproxy does not start on boot

M95D, 2024-10-30:
Show your /etc/config/tinyproxy (without passwords) and the output of `ls -la /etc/rc.d/*tinyproxy*`.

mal19992, 2024-10-30:
cat /etc/config/tinyproxy
```
config tinyproxy

option enabled 1
option User nobody
option Group nogroup
option Port 180
option Listen 192.168.3.93
option Timeout 600
option DefaultErrorFile "/usr/share/tinyproxy/default.html"
option StatFile "/usr/share/tinyproxy/stats.html"
#option LogFile "/var/log/tinyproxy.log"
option Syslog 1
option LogLevel Info
option MaxClients 100
option MinSpareServers 5
option MaxSpareServers 20
option StartServers 10
option MaxRequestsPerChild 30000
list Allow '127.0.0.1'
list Allow '192.168.3.0/24'
#list ConnectPort 443
option DisableViaHeader 'Yes'
option ViaProxyName 'tinyproxy'
option XTinyproxy 'No'

```
ls -la /etc/rc.d/*tinyproxy*
` /etc/rc.d/S95tinyproxy -> ../init.d/tinyproxy`
It used to be on level 50, I switched it to level 95 trying to fix. 
trying to debug S95tinyproxy
```
START=95
USE_PROCD=1
PROCD_DEBUG=1
export INIT_TRACE=1
export PROCD_DEBUG=1
```

It did not help. Only `sleep 60` helps

mal19992, 2024-10-30:
This is WAN (external) interface. I am not sure how late it gets up. It is initiated with DHCP on openwrt router boot, DHCP server is another locally installed router. Then the openwrt router with tinyproxy serves as a proxy for the people at the locally installed other router network, the reason -- I am not allowed to touch the main router, so some functionality is provided by an openwrt router on the same network. The openwrt  gets 192.168.3.93 as WAN IP from another  router.

M95D, 2024-10-30:
Yeah, it's probably caused by the late IF up. You should enable that log, to be sure.
If that's the reason, then it's not a bug. Proxy servers aren't supposed to bind to WAN IF. You have 3 choices:
- leave it like it is now, started with a delay, optionally disable service auto-start
- set the WAN to static IP, that way it will be up immediately
- remove the binding to WAN and let it bind to 0.0.0.0 instead

mal19992, 2024-10-31:
A web server should work over any interface. Other web servers work, only tinyproxy does not. There should be a starting program dependence/order, only tinyproxy does not have any. 

_then it's not a bug. Proxy servers aren't supposed ..._

Looks like a limited program unable to implement startup order dependence and/or crash handling (e.g. fail, restart, etc.)

M95D, 2024-10-31:
> A web server should work over any interface.

Except Tinyproxy is *not* a web server. It's a proxy server, and it's not even a *reverse*-proxy server! And, btw, no web server or any other program would specifically bind to an IF that isn't up. In fact, all web servers in their default config will bind 0.0.0.0, not a specific interface.

Like I said, a proxy it's not supposed to face the WAN, so it works as indended. If you need it to do something other than designed, write an IF-up hotplug script yourself. Or why don't you use a full Linux with systemd instead?

mal19992, 2024-10-31:
> no web server or any other program would specifically bind to an IF that isn't up

This is quite common for a web server. Two ways to solve (e.g. in systemd)
1. set startup dependence after network is up
2. in httpd.service.d/restart.conf add 
```
[Service]
Restart=always
RestartSec=10
```
which would restart the server on crash or failure to start every 10 seconds. I believe openwrt has something similar.

mal19992, 2024-10-31:
> Or why don't you use a full Linux with systemd instead?

I use tinyproxy on openwrt for space/extra equipment savement/etc. reasons. Correct, I can make apache to run on a desktop server,  But the beauty of openwrt -- a small device without cooling, strong power supply, etc. A very nice customizable device.