openwrt/packages #25212: sstp-client: unconditionally disables certificate validation

patrakov, 2024-10-27:
Maintainer: @vgaetera 
Environment: Any, for example, TP-Link Archer C7 v2 on OpenWrt 23.05.5

Description:

The `sstp.sh` protocol handler disables SSL certificate validation unconditionally by passing the `--cert-warn` option to `sstpc`. This is a security problem.

vgaetera, 2024-10-28:
Sorry, I'm not the maintainer of this package.
You should tag the actual maintainer.
See `PKG_MAINTAINER` in the `Makefile`.

patrakov, 2024-10-28:
Updated!

Neustradamus, 2024-12-09:
@enaess: What do you think about it?

mschwager, 2025-02-06:
Here are some other locations that may be disabling TLS verification:

- https://github.com/openwrt/packages/blob/1c1125a47e61422124bac6852676184be59fa33c/net/alist/files/alist.init#L82
- https://github.com/openwrt/packages/blob/1c1125a47e61422124bac6852676184be59fa33c/net/crowdsec-firewall-bouncer/files/crowdsec-firewall-bouncer.initd#L81
- https://github.com/openwrt/packages/blob/1c1125a47e61422124bac6852676184be59fa33c/net/nut/files/nut-monitor.init#L115

I found these by [searching the code for "verify"](https://github.com/search?q=repo%3Aopenwrt%2Fpackages%20verify&type=code). There may be some other more subtle variations in there where TLS verification is patched out of code rather than disabled by configuration.