openwrt/packages #25213: sstp-client: does not obey the interface metric

patrakov, 2024-10-27:
Maintainer: @vgaetera 
Environment: Any, for example, TP-Link Archer C7 v2 on OpenWrt 23.05.5

Description:

The `sstp.sh` protocol handler implements all option parsing from scratch instead of inheriting all common PPP option logic. This, for example, leads to the configured interface metric being ignored.

Example of an affected configuration:

```
config interface 'vpn'
	option proto 'sstp'
	option server 'sstp.vpn.example'
	option username 'username'
	option password 'password'
	option ipv6 '1'
	option log_level '0'
	option peerdns '0'
	list dns '8.8.8.8'
	list dns '8.8.4.4'
	option metric '91'
```

This results in the created route having no explicit metric, as opposed to metric 91.

A workaround exists, but this should not be necessary:
```
	option pppd_options 'defaultroute-metric 91'
```

Please look at how `pppossh` includes standard functions from `ppp.sh` and implements its protocol handler on top of that.

vgaetera, 2024-10-28:
Sorry, I'm not the maintainer of this package.
You should tag the actual maintainer.
See `PKG_MAINTAINER` in the `Makefile`.

Neustradamus, 2024-12-09:
@enaess: What do you think about it?