openwrt/packages #25223: ffmpeg: build fails with mold linker

M95D, 2024-10-29:
Maintainer: @thess, last updated by @krant
Environment: armv7, mvebu, Armada 385, Turris Omnia, OpenWrt git master/main
Description:
When CONFIG_USE_MOLD is selected, ffmpeg build fails with this error:
```
libavcodec/arm/mlpdsp_armv5te.S: Assembler messages:
libavcodec/arm/mlpdsp_armv5te.S:232: Error: junk at end of line, first unrecognized character is `0'
libavcodec/arm/mlpdsp_armv5te.S:367:  Info: macro invoked from here
libavcodec/arm/mlpdsp_armv5te.S:393:   Info: macro invoked from here
libavcodec/arm/mlpdsp_armv5te.S:414:    Info: macro invoked from here
libavcodec/arm/mlpdsp_armv5te.S:244: Error: garbage following instruction -- `bne 01b'
libavcodec/arm/mlpdsp_armv5te.S:367:  Info: macro invoked from here
libavcodec/arm/mlpdsp_armv5te.S:393:   Info: macro invoked from here
libavcodec/arm/mlpdsp_armv5te.S:414:    Info: macro invoked from here
```
Error repeats.

Is it possible to disable mold for selected packages, such as this one?
Thanks.

krant, 2024-10-30:
I've hit the same when tried to build it with newer binutils, it must be fixed in the next release - https://trac.ffmpeg.org/ticket/11074

M95D, 2024-10-30:
OK. I'll leave this open until the package is updated.

closed 2024-12-04 by M95D