openwrt/packages #25239: snort3 - official rules fail with ERROR: <<rulesfile>>:<<line>> unknown rule keyword: pcre.

sgofferj, 2024-10-31:
# Issue template (remove lines from top till here)

Maintainer: @flyn-org 
Environment: ARM64, Bananapi BPI-R4, OpenWrt SNAPSHOT (r27913-a2aabc9a7a)

Description:
When trying to start snort3, thousands of rules fail to load with `ERROR: <rulesfile>:<line no.> unknown rule keyword: pcre.`

`snort 3, 3.1.84.0-r4`
```
   ,,_     -*> Snort++ <*-
  o"  )~   Version 3.1.84.0
   ''''    By Martin Roesch & The Snort Team
           http://snort.org/contact#team
           Copyright (C) 2014-2024 Cisco and/or its affiliates. All rights reserved.
           Copyright (C) 1998-2013 Sourcefire, Inc., et al.
           Using DAQ version 3.0.15
           Using LuaJIT version 2.1.0-beta3
           Using OpenSSL 3.0.15 3 Sep 2024
           Using libpcap version 1.10.5 (with TPACKET_V3)
           Using PCRE version 10.42 2022-12-11
           Using ZLIB version 1.3.1
           Using LZMA version 5.6.2
```

Reproduce:
- Install snort
- Setup config with oinkcode
- Download official ruleset with snort-rules
- Restart snort

More info:
```
root@defiant:/tmp/snort.d/rules# ldd /usr/bin/snort
        /lib/ld-musl-aarch64.so.1 (0x7f80856000)
        libtirpc.so.3 => /usr/lib/libtirpc.so.3 (0x7f80815000)
        libdaq.so.3 => /usr/lib/libdaq.so.3 (0x7f807f4000)
        libdnet.so.1 => /usr/lib/libdnet.so.1 (0x7f807d3000)
        libhwloc.so.15 => /usr/lib/libhwloc.so.15 (0x7f80772000)
        libluajit-5.1.so.2 => /usr/lib/libluajit-5.1.so.2 (0x7f806f1000)
        libcrypto.so.3 => /usr/lib/libcrypto.so.3 (0x7f8031c000)
        libpcap.so.1 => /usr/lib/libpcap.so.1 (0x7f802bb000)
        libpcre2-8.so.0 => /usr/lib/libpcre2-8.so.0 (0x7f8025a000)
        libz.so.1 => /usr/lib/libz.so.1 (0x7f80229000)
        libatomic.so.1 => /lib/libatomic.so.1 (0x7f80207000)
        liblzma.so.5 => /usr/lib/liblzma.so.5 (0x7f801d6000)
        libuuid.so.1 => /usr/lib/libuuid.so.1 (0x7f801b5000)
        libtcmalloc.so.4 => /usr/lib/libtcmalloc.so.4 (0x7f7ffa2000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7f7fd9d000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f7fd6c000)
        libc.so => /lib/ld-musl-aarch64.so.1 (0x7f80856000)
        libpciaccess.so.0 => /usr/lib/libpciaccess.so.0 (0x7f7fd4b000)
        libunwind.so.8 => /usr/lib/libunwind.so.8 (0x7f7fd09000)
```

Logs not included because it's 12114 errors (463 warnings).

brada4, 2024-11-04:
One line triggering error?

sgofferj, 2024-11-05:
Random one:
```
2024-10-31 10:44:24.065 ERROR: /etc/snort/rules/snort3-server-webapp.rules:5197 unknown rule keyword: pcre.
```

```
alert tcp $EXTERNAL_NET any -> $HOME_NET $HTTP_PORTS ( msg:"SERVER-WEBAPP Advantech WISE-PaaS RMM UpgradeMgmt upload_ota directory traversal attempt"; flow:to_server,established; http_uri; content:"/UpgradeMgmt/upload_ota",fast_pattern,nocase; content:"frmUpdateSetting_UploadFileFullName=",nocase; content:"../"; pcre:"/[?&]frmUpdateSetting_UploadFileFullName=[^&]*?\x2e\x2e\x2f/i"; metadata:policy max-detect-ips drop,policy security-ips drop; service:http; reference:cve,2019-13551; classtype:web-application-attack; sid:59319; rev:1; )
```

graysky2, 2024-11-11:
Not sure why you're seeing this but try pulling https://github.com/openwrt/packages/pull/25267 (and see comments in the PR)

sgofferj, 2024-11-12:
@graysky2 
I'm neither familiar enough with OpenWRT yet, nor with snort to build this. I also only have this one production box and I'd rather not have all build tools on a production firewall.

sgofferj, 2024-11-13:
After reading a little bit, I ran sed -i "s/pcre/pcre2/g" over all rule files and snort starts and seems to work. At least I'm getting alert. However, that is not in the documentation. I think, it would be good to have this in the documentation or better, have snort-rules do this automatically and transparently to the user.

graysky2, 2024-11-13:
The sed line is only temporary as upstream reworks the PR and implements, see: https://github.com/snort3/snort3/pull/326#issuecomment-2462663172