openwrt/packages #25244: glib2: Bulding OpenWrt v23.05.5 with glib2 fails at building glib2 for inadequate Meson version

risa2000, 2024-11-03:
Maintainer: @yegorich @neheb
Environment:
Build system: Debian 12.7 (arm64)
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc' 
DEVICE=tplink_archer-ax23-v1

Description:
When building OpenWrt from source (git tag v23.05.05) with glib2 package, the build fails with error:
```log
make[3]: Entering directory '/home/risa/work/openwrt-build-env/openwrt/feeds/packages/libs/glib2'
rm -f /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/.configured_*
rm -f /home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.glib2_installed
/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'gcc',|" -e "s|@CXX@|'g++',|" -e "s|@PKGCONFIG@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-O2', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/hostpkg/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/host/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/hostpkg/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/host/include',|" -e "s|@CXXFLAGS@|'-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/hostpkg/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/host/include',|" -e "s|@LDFLAGS@|'-L/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/lib', '-L/home/risa/work/openwrt-build-env/openwrt/staging_dir/hostpkg/lib', '-L/home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/host/lib',|" -e "s|@PREFIX@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/hostpkg|" < /home/risa/work/openwrt-build-env/openwrt/staging_dir/host/lib/meson/openwrt-native.txt.in > /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-native.txt
/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/sed -e "s|@CC@|'mipsel-openwrt-linux-musl-gcc',|" -e "s|@CXX@|'mipsel-openwrt-linux-musl-g++',|" -e "s|@AR@|mipsel-openwrt-linux-musl-gcc-ar|" -e "s|@STRIP@|mipsel-openwrt-linux-musl-strip|" -e "s|@NM@|mipsel-openwrt-linux-musl-gcc-nm|" -e "s|@PKGCONFIG@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/pkg-config|" -e "s|@CMAKE@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/cmake|" -e "s|@PYTHON@|/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/python3|" -e "s|@CFLAGS@|'-Os', '-pipe', '-mno-branch-likely', '-mips32r2', '-mtune=24kc', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-msoft-float', '-fmacro-prefix-map=/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0=glib-2.82.0', '-mips16', '-minterlink-mips16', '-ffunction-sections', '-fdata-sections', '-Wformat', '-Werror=format-security', '-fstack-protector', '-Wl,-z,now', '-Wl,-z,relro', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/usr/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/include/fortify', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/include',|" -e "s|@CXXFLAGS@|'-Os', '-pipe', '-mno-branch-likely', '-mips32r2', '-mtune=24kc', '-fno-caller-saves', '-fno-plt', '-fhonour-copts', '-msoft-float', '-fmacro-prefix-map=/home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0=glib-2.82.0', '-mips16', '-minterlink-mips16', '-ffunction-sections', '-fdata-sections', '-Wformat', '-Werror=format-security', '-fstack-protector', '-Wl,-z,now', '-Wl,-z,relro', '-mips16', '-minterlink-mips16', '-ffunction-sections', '-fdata-sections', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/usr/include', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/include/fortify', '-I/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/include',|" -e "s|@LDFLAGS@|'-L/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/usr/lib', '-L/home/risa/work/openwrt-build-env/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-13.1.0_musl/lib', '-Wl,--gc-sections', '-znow', '-zrelro',|" -e "s|@ARCH@|"mips"|" -e "s|@CPU@|"generic"|" -e "s|@ENDIAN@|little|" < /home/risa/work/openwrt-build-env/openwrt/staging_dir/host/lib/meson/openwrt-cross.txt.in > /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-cross.txt
/home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/python3 /home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/meson.py  --buildtype plain --native-file /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-native.txt --cross-file /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-cross.txt -Dglib_debug=disabled -Dlibmount=disabled -Dselinux=disabled -Dglib_assert=false -Dtests=false -Dintrospection=disabled -Druntime_dir=/var/run -Dxattr=true -Db_lto=true -Dnls=disabled /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-build /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-build/..
The Meson build system
Version: 0.61.5
Source dir: /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0
Build dir: /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-build
Build type: cross build

../../../../build_dir/target-mipsel_24kc_musl/glib-2.82.0/meson.build:1:0: ERROR: Meson version is 0.61.5 but project requires >= 1.2.0

A full log can be found at /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-build/meson-logs/meson-log.txt
make[3]: *** [Makefile:136: /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/.configured_efb2d3cf01c707184fb77fd0fe149f94] Error 1
make[3]: Leaving directory '/home/risa/work/openwrt-build-env/openwrt/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.44#0.01#0.45
    ERROR: package/feeds/packages/glib2 failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/glib2/compile] Error 1
make[2]: Leaving directory '/home/risa/work/openwrt-build-env/openwrt'
make[1]: *** [package/Makefile:123: /home/risa/work/openwrt-build-env/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/risa/work/openwrt-build-env/openwrt'
make: *** [/home/risa/work/openwrt-build-env/openwrt/include/toplevel.mk:233: world] Error 2
```
The file mentioned in the log for the full log does not show much:
```
Build started at 2024-11-03T10:38:02.746942
Main binary: /home/risa/work/openwrt-build-env/openwrt/staging_dir/host/bin/python3
Build Options: -Dglib_debug=disabled -Dlibmount=disabled -Dselinux=disabled -Dglib_assert=false -Dtests=false -Dintrospection=disabled -Druntime_dir=/var/run -Dxattr=true -Db_lto=true -Dnls=disabled -Dbuildtype=plain '--cross-file /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-cross.txt' '--native-file /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-native.txt'
Python system: Linux
The Meson build system
Version: 0.61.5
Source dir: /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0
Build dir: /home/risa/work/openwrt-build-env/openwrt/build_dir/target-mipsel_24kc_musl/glib-2.82.0/openwrt-build
Build type: cross build

../../../../build_dir/target-mipsel_24kc_musl/glib-2.82.0/meson.build:1:0: ERROR: Meson version is 0.61.5 but project requires >= 1.2.0
```
First I thought I should install right Meson, but after I did, I realized, it does not use Python from the OS, but from the build environment.

robimarko, 2024-11-03:
Are you sure that you are using openwrt-23.05 branch for packages?

Because, 23.05 has glib2 2.74.7 and not 2.82.0

risa2000, 2024-11-03:
> Are you sure that you are using openwrt-23.05 branch for packages?
> Because, 23.05 has glib2 2.74.7 and not 2.82.0

@robimarko
No, I am not sure. My routine is basically this:
```bash
cd openwrt
git checkout v23.05.5
make distclean
git clean -df
scripts/feeds update
scripts/feeds install <packages I use in my image>
cp my_config .config
make oldconfig
make -j4
```
I do not know, how should I control which packages I should use, I would have thought if there is such a dependency, it would be controlled by the openwrt commit currently checked-out?

robimarko, 2024-11-03:
Go into `feeds/packages` and check with `git status` or `git log` on which branch you currently are.

Also, there is no point in installing just the packages you use, install all of them with `-a`.
You can replace the individual feeds update and then install calls just with one `make package/symlinks`

risa2000, 2024-11-03:
My `feeds/packages` seem to be on the `master` branch.
```
git log
commit d33977e50de97c5697eb92aad90db3dec5daf902 (grafted, HEAD -> master, origin/master, origin/HEAD)
```

For what concerns `feeds.conf.default` mine reads:
```
src-git packages https://git.openwrt.org/feed/packages.git^b5ed85f6e94aa08de1433272dc007550f4a28201
src-git luci https://git.openwrt.org/project/luci.git^63ba3cba5b7bfb803a875d4d8f01248634687fd5
src-git routing https://git.openwrt.org/feed/routing.git^e351d1e623e9ef2ab78f28cb1ce8d271d28c902d
src-git telephony https://git.openwrt.org/feed/telephony.git^98c8a5aa4624312ed758e2e2b6d4039050a1649d
```
Which I guess is correct, but what I did is actually override it with this (because I need `oldpackages` and do not need some others) when building the current `master` for `openwrt`:
```
src-git packages https://git.openwrt.org/feed/packages.git
src-git routing https://git.openwrt.org/feed/routing.git
src-git oldpackages http://git.openwrt.org/packages.git
```
Actually I had it this way all the time and never realized there could be a mismatch. To my defense, I always built the  `master` until now. The reason I went for the latest release was some issues I see in the current build and wanted to compare it to the latest stable.
Now I see my problem! Thanks for pointing this out 👍 

I changed the `packages` and `routing` feeds in my `feeds.conf` to the specific commit suggested in `feed.conf.default` and left `oldpackages` on `master` and now `glib2` builds fine, but `ntpd` does not build. So I guess I need to reference "right" commit from `oldpackages` as well. Do you know, how could I find the right one?

For what concerns installing only used packages, I do it because I only add few packages to the vanilla config and my understanding was that once the package is installed it is picked up by the config and stored there as well. Having all packages in the config will be only a distraction when trying to figure out what has changed from version to version, or commit to commit, because I do not need to track them.

robimarko, 2024-11-04:
I dont understand what is the point of so-called `oldpackages`, but do not mix different branches when building.

I think that ntpd is just broken on 23.05 branch:
https://github.com/openwrt/packages/pull/24921

For handling the config, just use diffconfig, then you will only get the diff on top of defconfig and not the whole config.

risa2000, 2024-11-04:
> I dont understand what is the point of so-called `oldpackages`, but do not mix different branches when building.

`ntpd` is in `oldpackages`. I just need to figure out, which commit is compatible with `v23.05.5`. I guess I will try going by the date first.

risa2000, 2024-11-05:
It turned out that `ntpd` indeed did not build with `v23.05.5`, so I had to figure out a different `openwrt` commit which went along with all the packages to build an alternative.
I guess I understand why `oldpackages` are not supported now.
This issue can be closed.

closed 2024-11-05 by risa2000