openwrt/packages #25248: openvpn-easy-rsa: Update to 3.2.1

Neustradamus, 2024-11-04:
Maintainer: @luizluca, @neheb

Environment: ALL

Description:
```
Update openvpn-easy-rsa to 3.2.1
```

Thanks in advance.

Links:
- https://github.com/openwrt/packages/tree/master/net/openvpn-easy-rsa
- https://github.com/OpenVPN/easy-rsa
- https://github.com/OpenVPN/easy-rsa/releases
- https://easy-rsa.readthedocs.io/en/latest/

luizluca, 2024-11-07:
I migrated all my VPN to wireguard. I didn't noticed it was broken. Maybe we need someone else to take this pkg. However, I might take a look on it.

Neustradamus, 2024-11-19:
@luizluca: Have you tested?

It has been merged here:
- https://github.com/openwrt/packages/pull/25310

GrumpyMeow, 2025-01-10:
This package is currently already bumped to v3.2.1. This issue should be closed.

closed 2025-01-10 by Neustradamus