openwrt/packages #25266: ntpd compilation fails

wwortel, 2024-11-07:
ntpd
I need ntpd in my images but compile of any OpenWrt image fails at ntpd.
Have also tried p18 which is at ntpd' source but same problem. 'make V=s' shows:
[ ! -d ./src/ ] || cp -fpR ./src/. /var/sdks/OpenWrt/master/build_dir/target-mipsel_24kc_musl/ntp-4.2.8p17

Applying /var/sdks/feeds/packages/net/ntpd/patches/0001-libntp-Do-not-use-PTHREAD_STACK_MIN-on-glibc.patch using plaintext: 
patching file libntp/work_thread.c
Hunk #1 FAILED at 41.
1 out of 1 hunk FAILED -- saving rejects to file libntp/work_thread.c.rej
Patch failed!  Please fix /var/sdks/feeds/packages/net/ntpd/patches/0001-libntp-Do-not-use-PTHREAD_STACK_MIN-on-glibc.patch!
[work_thread.c.rej.txt](https://github.com/user-attachments/files/17659703/work_thread.c.rej.txt)

brada4, 2024-11-10:
Please fill the issue template incl diffconfig.

brada4, 2024-11-10:
particular patch applies to ntpd .... p15 and EOL-d v22. If you are experimenting in archaeology it is up to you to backport (in this case delete) patches.
Or use stable version.

wwortel, 2024-11-10:
Mmm, experimenting archeology?? After getting stuck I actually did a completely fresh install of OpenWrt, and just maintained links to 'feeds' and 'dl' folders, but updated all packages using  './scripts/feeds update -a'
Have not touched ntpd at all, except its make file but only after having stumbled over this compile error to test the most recent v18 version, with same problem. So, judging from your comment, the packages update seems not to have produced a coherent situation between patches and ntpd source code.

closed 2024-11-10 by wwortel

brada4, 2024-11-10:
You can click release branches in github and literarily compare file listings and git log to ones on disk for breaking package. Also check faillogs on download server if no problem pops up in first check.