openwrt/packages #25276: cgi-io:  Reading part data, byte offset 199489, the value exceeds the maximum allow size.

peterwillcn, 2024-11-08:
At reading part data, byte offset 199489, the value exceeds the maximum allow size
In mimedecode_message_body(), file /usr/share/ucode/luci/http.uc, line 112, byte 11:
  called from function parse_message_body (/usr/share/ucode/luci/http.uc:180:51)
  called from function [anonymous function] (/usr/share/ucode/luci/http.uc:450:3)
  called from function [anonymous function] (/usr/share/ucode/luci/http.uc:374:22)
  called from function require_post_security (/usr/share/ucode/luci/dispatcher.uc:680:47)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1000:35)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `            die(err);`
  Near here ---------^

brada4, 2024-11-10:
How did you get there?

jow-, 2025-03-26:
This is not related to `cgi-io` but to the multipart parser of LuCI (liblucihttp): https://github.com/jow-/lucihttp/blob/master/lib/multipart-parser.c#L537-L545

It means one of your form values is larger than the limit passed to the message parser. By default, individual field contents must not exceed 100KB in size iirc. You could try increasing `luci.http.HTTP_MAX_CONTENT` in your controller before accessing the first field

closed 2025-03-26 by jow-