openwrt/packages #25277: nut-cgi Only shows Authorization Required message

cconde, 2024-11-08:
# Issue template (remove lines from top till here)

Maintainer: @\neheb
Environment: UBNT_E300 (CN7030p1.2-1000-AAP) / Ubiquiti EdgeRouter 4 / OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-23.093.57360-e98243e

Description:

After proper configuration of nut-monitor I installed nut-CGI, so I can now access my router /nut webpage, properly loading and showing stats and configuration links.

[image]

When accessing any of them, only a black page with "Authorization required" is shown.

I don't know what to do as the logs show that nut-monitor is properly connected to remote UPS (with authorization) and [nut-cgi](https://openwrt.org/docs/guide-user/services/ups/software.nut#nut_cgi) doesn't have any property for authorization.

What may be I missing?

Thanks!

brada4, 2024-11-10:
Your device is perfectly supported by latest OpenWRT 23.05.5 , if you still encounter problems after upgrade feel free to adjust your post with supportable numbers.

cconde, 2024-11-14:
Afternupgrading I'm getting this error when clicking on menu option:

```
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/model/cbi/nut_server.lua:133: attempt to call a nil value

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:817:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:818:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1003:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

         return lcall.call(modname, method, ...args);
  Near here ----------------------------------------^
```

I tried uninstalling and reinstalling luci-app-nut (forcing files) but no luck.

Maybe it's related to a leftover, but I'd need a little guidance to ensure everything is clean before retrying.

brada4, 2024-11-14:
Not related to initial issue, approach forums.

cconde, 2024-11-15:
Approached and fixed, thanks, now on OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.313.38121-76e4eca

Error is still there, I just get an Authrization Error when browsing:
http://myDevice/cgi-bin/nut/upsstats.cgi
http://myDevice/cgi-bin/nut/upsset.cgi

The UPS client seems to be working properly and connect to the remote UPS network server

brada4, 2024-11-15:
Web log would be nice to add..

pharmacologic, 2025-06-05:
I was also having trouble with this, and found this issue in my searches. I had been trying the username/password combination(s) which I had just configured in `/etc/config/nut_server` for the basic authentication when trying to access nut-cgi, which consistently failed.

I finally just tried my openwrt login for it and that's what actually worked. 

So I think you just need to re-authenticate with openwrt to access nut-cgi (maybe it doesn't use the same cookies?)

danielfdickinson, 2025-12-14:
@nx6 There isn't from the GUI (that I recall), but if you edit `/etc/httpd.conf` you can use a user:password of your choosing (not sure what hashes are available for that file though, these days, as it's a `uhttpd` that was used for legacy reasons when `uhttpd` was relatively new.