openwrt/packages #25315: On Teltonika RUTx devices, the init script fails due to failure to resolve interface name

luckman212, 2024-11-12:
To get darkstat to initialize with any interface other than `'lan'` (e.g. `qmimux0`), I had to change
 https://github.com/openwrt/packages/blob/d6ad08828d9ef3d740456a227293565fff8cc99c/net/darkstat/files/darkstat.init#L38-L41

to

```sh
build_config_string() {
       local cfg="$1"                                  
       config_get interface $cfg interface 
       network_get_device ifname "$interface" || ifname="$interface"
```

```
# darkstat --version
darkstat 3.0.719 (using libpcap version 1.9.1 (with TPACKET_V3))
```

Teltonika RUTX11, firmware RUTX_R_00.07.10.2 based on OpenWrt 21.02

luckman212, 2024-11-21:
Any comments on this? Is it worth submitting an actual PR to get this change mainlined?

brada4, 2024-12-14:
This is not teltonika bug tracker.

luckman212, 2024-12-14:
Okay. Sorry I thought maybe this was an upstream bug.