openwrt/packages #25338: git: Image Builder Error

shanpo, 2024-11-13:
```
ERROR: git-2.46.2-r1: usr/lib/git-core/git: no hard link target (git) in archive
ERROR: git-2.46.2-r1: usr/lib/git-core/git-shell: no hard link target (git-shell) in archive
```

xun188, 2024-11-15:
Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
fetch https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/packages.adb
fetch https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.60-1-a59ba0c2e57510a4215b673a9b7cba9f/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/packages.adb
WARNING: opening packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: git-2.46.2-r1: usr/lib/git-core/git: no hard link target (git) in archive
ERROR: git-2.46.2-r1: usr/lib/git-core/git-shell: no hard link target (git-shell) in archive
make[2]: *** [Makefile:229: package_install] Error 1
make[1]: *** [Makefile:170: _call_manifest] Error 2
make: *** [Makefile:344: manifest] Error 2

shanpo, 2024-11-16:
commit:[ba445ac](https://github.com/openwrt/packages/commit/ba445acd7cbd39375a0ed382896b6236a65bd780)Solved problem

closed 2024-11-16 by shanpo