openwrt/packages #25348: Libcbor: does not compile: ends with "cannot find -lasan/-lubsan"

pwned-pixel, 2024-11-14:
PKG_MAINTAINER:= Linos Giannopoulos <linosgian00+openwrt@gmail.com>
Maintainer: cannot find any github related name (so I've used the above)
Environment: OpenWrt SNAPSHOT as of today with commit 0e59eaa / Packages as of today with commit 78ccd86
Host system: ArchLinux (latest)

Libcbor compile ends with:

```
/home/pwned/nolf/openwrt/tchain14/toolchain/bin/../lib/gcc/aarch64-openwrt-linux-musl/14.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld.bfd: cannot find -lasan: No such file or directory
/home/pwned/nolf/openwrt/tchain14/toolchain/bin/../lib/gcc/aarch64-openwrt-linux-musl/14.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld.bfd: cannot find -lubsan: No such file or directory
```

Disabling the bug detector is "fixing" it:

```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -55,7 +55,7 @@
     add_definitions(-DPRINT_FUZZ)
 endif(PRINT_FUZZ)
 
-option(SANITIZE "Enable ASan & a few compatible sanitizers in Debug mode" ON)
+option(SANITIZE "Enable ASan & a few compatible sanitizers in Debug mode" OFF)
 
 set(CPACK_GENERATOR "DEB" "TGZ" "RPM")
 set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
```

I've tested gcc14 only. No clue if it hits on gcc13 as well. Let me know if I have to test.
Came across this by accidentally enabling libfido2 support for openssh-server. It is pulling this lib.
If more output is needed let me know. There are some unrelated warnings (e.g. "Policy CMP0054 is not set" and "-Wreturn-type errors within src/cbor.c and src/cbor/serialization.c and src/cbor/ints.c) also.

pwned-pixel, 2024-11-27:
I'm closing this for now to not have it like many other issues lingering around endlessly.

It seems to me this package is not used by a wider audience resp. there is no active maintainer available.
As there is a workaround for this issue it isn't urgent anyway. If there is a new maintainer available he/she would propably stumble over this issue too (if it would still exist).

closed 2024-11-27 by pwned-pixel

hnyman, 2025-04-13:
I suddenly ran into this bug in one of my builds, and strangely enough, I can't even figure out why libcbor is now being compiled in. It is not enabled in .config, and I can't even find any reference to it in build_dir/target... dir files. Strange.

(To my knowledge this libcbor has never been included in any of my builds.)

pwned-pixel, 2025-04-14:
Now as you are saying this: 

> I can't even figure out why libcbor is now being compiled in

For me in the past I had the same impression. I remember I've made some changes within menuconfig (what I don't do usually; just using my seed config) and I thought I've enabled libfido2 by accident. I cannot remember exactly what I did test. But I was sure I never touched libfido2. But I cannot rule it out. Maybe I've enabled and disabled sth. related to openssh and this on/off was enough to pull this libs in. Sadly I cannot remember if disabling libfido2 was enough to stop compiling it or if I just started over with a new config. I assume the later was the case.

PR #26317 (2025-04-14): libcbor: disable sanitize by default
Maintainer: @linosgian
Compile tested: rockchip/armv8
Run tested: n/a

Description:
musl libc does not support this feature, and for glibc, additional
library libasan is required. Disable it explicitly to avoid build issue
when CONFIG_DEBUG is selected.

Fixes: #25348
Cc: @hnyman @pwned-pixel