openwrt/packages #25371: adguardhome uninstallable with apk

lightsabata, 2024-11-15:
**Maintainer :** @dobo90 
**Environment :** ARM Corex-A73 (MediaTek MT7988)
**OpenWrt version :** SNAPSHOT r28072-b902284290 / LuCI Master 24.315.52043~d650688

**Description :**
Since last snapshot and the depreciation of opkg, i am unable to install the package adguardhome.
i guess the package need some minor adjustements.
Thanks

```
root@ROUTER:~# apk add adguardhome
ERROR: unable to select packages:
  luci-app-package-manager-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-24.315.52043~d650688[luci-app-package-manager]
  luci-app-firewall-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-light-24.315.52043~d650688[luci-app-firewall]
  luci-mod-network-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-mod-admin-full-24.315.52043~d650688[luci-mod-network]
  luci-mod-system-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-mod-admin-full-24.315.52043~d650688[luci-mod-system]
  luci-mod-admin-full-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-light-24.315.52043~d650688[luci-mod-admin-full]
  luci-proto-ipv6-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-light-24.315.52043~d650688[luci-proto-ipv6]
  luci-proto-ppp-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-light-24.315.52043~d650688[luci-proto-ppp]
  luci-theme-bootstrap-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-light-24.315.52043~d650688[luci-theme-bootstrap]
  luci-light-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-24.315.52043~d650688[luci-light]
  luci-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: world[luci]
  luci-lib-base-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: luci-lua-runtime-24.315.52043~d650688[luci-lib-base]
  luci-compat-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: world[luci-compat]
  luci-lib-ipkg-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: world[luci-lib-ipkg]
  luci-theme-material-24.315.52043~d650688:
    error: uninstallable
    arch: all
    satisfies: world[luci-theme-material]
```

Ra2-IFV, 2024-11-17:
It's not related to the package

lightsabata, 2024-11-20:
yes after latest snapshot wich is reset all insdtalled package, i am able to install adguardhome, i appologise for the error on my side

closed 2024-11-20 by lightsabata