openwrt/packages #25372: nfs-kernel-server not available using apk yet nfs-kernel-server-2.6.2-r4.apk  exists

dynasticorpheus, 2024-11-15:
Maintainer : Peter Wagner <tripolar@gmx.at>
Environment : arm_cortex-a9_vfpv3-d16
OpenWrt version : OpenWrt SNAPSHOT r28085-6720c4ccba / LuCI Master 24.319.72217~9fe1e34

When trying to install nfs-kernel-server via apk it reports not found however [nfs-kernel-server-2.6.2-r4.apk](https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/nfs-kernel-server-2.6.2-r4.apk) exists and works when manually downloaded / installed.

```
root@OpenWrt:~# apk update
fetch https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/packages.adb
fetch https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/6.6.60-1-4e67c16a61d298c224f10ead7bb7c639/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/routing/packages.adb
fetch https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/telephony/packages.adb
 [https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/base/packages.adb]
 [https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/kmods/6.6.60-1-4e67c16a61d298c224f10ead7bb7c639/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/luci/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/routing/packages.adb]
 [https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/telephony/packages.adb]
OK: 10267 distinct packages available

root@OpenWrt:~# apk search nfs
kmod-fs-nfs-6.6.60-r1
kmod-fs-nfs-common-6.6.60-r1
kmod-fs-nfs-common-rpcsec-6.6.60-r1
kmod-fs-nfs-v3-6.6.60-r1
kmod-fs-nfs-v4-6.6.60-r1
kmod-fs-nfsd-6.6.60-r1
nfs-utils-2.6.2-r4
nfs-utils-libs-2.6.2-r4

root@OpenWrt:~# apk search nfs-kernel-server
root@OpenWrt:~#
```

Ra2-IFV, 2024-11-17:
Looks like an indexing problem... How about trying again later?

efahl, 2024-11-20:
Found it, any package names containing `kernel-` (or `base-files-` or `libc-`) are being filtered out of the package indexes.  Just needs an anchor `^` at the front of the grep patterns.

https://github.com/openwrt/openwrt/blob/main/package/Makefile#L134

(@robimarko you seem to be poking around in this stuff...)

closed 2024-11-21 by openwrt-bot

efahl, 2024-11-21:
We're getting there, slowly but surely!  👍🏼