openwrt/packages #25376: [BanIP]: user.err banIP: no download utility with SSL support

JuliusBairaktaris, 2024-11-15:
Maintainer: @dibdot
Environment: OpenWrt SNAPSHOT 749a43325b98afe4430fa7e2f23ac3ee44791ad6

Description:

With the new APK transition, BanIP no longer has the necessary dependencies to download files via HTTPS.

My config file: https://github.com/JuliusBairaktaris/Qualcommax_NSS_Builder/blob/main/ax3600.config

I ran `make defconfig` to generate the full .config

dibdot, 2024-11-15:
It's a known issue with newly introduced apk package manager in master and a workaround exists, see https://github.com/openwrt/openwrt/issues/16907#issuecomment-2478298115 for details.

closed 2024-11-15 by dibdot

JuliusBairaktaris, 2025-01-18:
Hey @dibdot, I just clean flashed a build with BanIP 1.5 and the issue reappeared.

dibdot, 2025-01-18:
worksforme, make sure you also have the appropriate SSL/TLS library installed (see readme)... or use curl or wget-ssl.

JuliusBairaktaris, 2025-01-18:
[image]

Its installed. Curl works however