openwrt/packages #25387: Ntpd compilation failure

kd1995-max, 2024-11-16:
Hello there,
I attempted to build my OpenWRT image with `ntpd` version 4.2.8p17, but the build is failing. After reviewing the logs, I couldn't identify the specific reason for the failure.
```
dev@dev-HP-15-Notebook-PC:~/openwrt$ sudo make package/feeds/packages/ntpd/compile V=s
....
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking for vfork.h... no
checking for sys/time.h... yes
checking for stdarg.h... yes
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking how to run the C preprocessor... aarch64-openwrt-linux-musl-gcc -E
checking for grep that handles long lines and -e... /home/dev/openwrt/staging_dir/host/bin/grep
checking for egrep... /home/dev/openwrt/staging_dir/host/bin/grep -E
checking whether aarch64-openwrt-linux-musl-gcc needs -traditional... no
checking if aarch64-openwrt-linux-musl-gcc can handle #warning... yes
checking if aarch64-openwrt-linux-musl-gcc supports __attribute__((...))... yes
checking if aarch64-openwrt-linux-musl-gcc can handle -Wstrict-overflow... yes
checking if aarch64-openwrt-linux-musl-gcc can handle -Winit-self... yes
checking if aarch64-openwrt-linux-musl-gcc can handle -Wno-format... yes
checking if linker supports omitting unused code and data... no
checking additional compiler flags... none needed
checking additional linker flags... none needed
checking whether byte ordering is bigendian... (cached) no
checking for working volatile... yes
checking how to run the C preprocessor... aarch64-openwrt-linux-musl-gcc -E
checking for compile/link hardening flags... in file harden/linux
checking for install dir and man conventions... in file loc/legacy
checking for aarch64-openwrt-linux-ar... aarch64-openwrt-linux-musl-gcc-ar
checking the archiver (aarch64-openwrt-linux-musl-gcc-ar) interface... ar
checking for a sed that does not truncate output... /home/dev/openwrt/staging_dir/host/bin/sed
checking for fgrep... /home/dev/openwrt/staging_dir/host/bin/grep -F
checking how to print strings... printf
checking for pid_t... yes
checking for fork... yes
checking for vfork... yes
checking for vprintf... yes
checking for size_t... yes
checking for aarch64-openwrt-linux-musl-gcc options needed to detect all undeclared functions... none needed
checking whether strerror_r is declared... yes
checking for unsigned long long int... yes
checking for vsnprintf... (cached) yes
checking whether vsnprintf is C99 compliant... no
checking if C99-snprintf replacement vsnprintf will be used... yes
checking for inttypes.h... (cached) yes
checking for locale.h... no
checking for stddef.h... no
checking for stdint.h... (cached) yes
checking for struct lconv.decimal_point... yes
checking for struct lconv.thousands_sep... yes
checking for long double... yes
checking for long long int... yes
checking for unsigned long long int... (cached) yes
checking for size_t... (cached) yes
checking for intmax_t... no
checking for uintmax_t... no
checking for uintptr_t... no
checking for ptrdiff_t... no
checking for localeconv... yes
checking for an ANSI C-conforming const... yes
checking for snprintf... yes
checking whether snprintf is C99 compliant... no
checking if C99-snprintf replacement snprintf will be used... yes
checking for ld used by aarch64-openwrt-linux-musl-gcc... aarch64-openwrt-linux-musl-ld
checking if the linker (aarch64-openwrt-linux-musl-ld) is GNU ld... yes
checking for directory path separator... '/'
checking line editing libraries... no
checking for library containing cos... none required
checking for working fork... cross
configure: WARNING: result yes guessed because of cross compilation
checking for working vfork... (cached) yes
checking for working alloca.h... yes
checking for alloca... yes
checking whether strerror_r returns char *... no
checking for library containing clock_gettime... none required
checking for clock_getres... yes
checking for clock_gettime... yes
checking for clock_settime... yes
checking for getclock... no
checking for stime... yes
checking for timegm... yes
checking for strlcpy... yes
checking for strlcat... yes
checking for library containing snprintb... no
checking for errno.h... no
checking for sys/socket.h... no
checking for sys/types.h... (cached) yes
checking for time.h... no
checking for net/if.h... yes
checking for linux/if_addr.h... yes
checking for arpa/nameser.h... no
checking for sys/param.h... no
checking for sys/time.h... (cached) yes
checking for sys/timers.h... no
checking for sys/sysctl.h... no
checking for netinet/in_system.h... no
checking for netinet/in_systm.h... no
checking for netinet/in.h... no
checking for resolv.h... no
checking for net/if_var.h... no
checking for netinet/ip.h... no
checking for netinet/in_var.h... no
checking for library containing inet_pton... none required
checking for library containing inet_ntop... none required
checking for library containing setsockopt... none required
checking for recvmsg... yes
checking for inline... inline
checking size of time_t... 0
checking whether char is unsigned... yes
checking size of signed char... 0
checking for s_char... no
checking for long long... no
checking size of short... (cached) 2
checking size of int... (cached) 4
checking size of long... (cached) 8
configure: error: No way to specify a signed character!
make[5]: *** [Makefile:717: config.status] Error 1
make[5]: Leaving directory '/home/dev/openwrt/build_dir/target-aarch64_cortex-a53_musl/ntp-4.2.8p17/sntp'
make[4]: *** [Makefile:665: all-recursive] Error 1
make[4]: Leaving directory '/home/dev/openwrt/build_dir/target-aarch64_cortex-a53_musl/ntp-4.2.8p17'
make[3]: *** [Makefile:597: all] Error 2
make[3]: Leaving directory '/home/dev/openwrt/build_dir/target-aarch64_cortex-a53_musl/ntp-4.2.8p17'
make[2]: *** [Makefile:162: /home/dev/openwrt/build_dir/target-aarch64_cortex-a53_musl/ntp-4.2.8p17/.built] Error 2
make[2]: Leaving directory '/home/dev/openwrt/feeds/packages/net/ntpd'
time: package/feeds/packages/ntpd/compile#167.24#20.79#187.74
    ERROR: package/feeds/packages/ntpd failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/ntpd/compile] Error 1
make[1]: Leaving directory '/home/dev/openwrt'
make: *** [/home/dev/openwrt/include/toplevel.mk:233: package/feeds/packages/ntpd/compile] Error 2
```

brada4, 2024-11-16:
make clean first then compile again.

kd1995-max, 2024-11-17:
I did, but didn't worked

brada4, 2024-11-17:
I did and it worked, both 23.05 and SNAPSHOT.
wipe and disable ccache, get rid of 3rd party repos and do not change a single checkbox past target model and enabling ntpd.

kd1995-max, 2024-11-17:
thanks @brada4 it worked bro 
I freshly cloned openwrt and then enabled ntpd
it worked!!!
Thank a ton bro

closed 2024-11-17 by kd1995-max

sidepipe, 2024-11-19:
Sorry but I've been having the same issue - essentially sntp fails to compile... this is on Ubuntu 24.10 and using OpenWrt 23.05.5 cloned from source. I'm not sure that deleting everything and starting again is really a solution!

On further investigation, it seems to be because the build process involves running ./config.status --recheck in the sntp subdirectory, which re-runs the configure script but without the environment variables set ( so important ones such as hw_cv_func_vsnprintf_c99 aren't set, and in this example this variable gets set to 0 because that's the default in the configure script if cross compiling is detected ).

I've "solved" this by adding a configure section to the ntpd Makefile, thus:

    define Build/Configure
        $(call Build/Configure/Default, $(CONFIGURE_VARS))
    endef

Presumably this works because the variables are embedded into config.status and therefor handled as arguments. However, this doesn't explain WHY this happens, and why it might be that most of the time it seems to work ( well, I've never got it to work on Ubuntu 24.10, but really haven't had many attempts and not tried compilation with a clean .config because time is limited, but I've never had an issue in the past and have compiled OpenWrt MANY times ).

Any ideas as to the root cause? In the meantime hopefully this workaround may help if someone else has the issue.

brada4, 2024-11-19:
Cause is basic compiler diagnostics by means of autotools failing. If you are after it - delete ccache data and try to compile, then delete any autotools cache and try again. In short log it is not visible, but should be same unsigned char failing somewhere before in full cleaned build log.

edgeinfinity1, 2024-12-24:
> Sorry but I've been having the same issue - essentially sntp fails to compile... this is on Ubuntu 24.10 and using OpenWrt 23.05.5 cloned from source. I'm not sure that deleting everything and starting again is really a solution!
> 
> On further investigation, it seems to be because the build process involves running ./config.status --recheck in the sntp subdirectory, which re-runs the configure script but without the environment variables set ( so important ones such as hw_cv_func_vsnprintf_c99 aren't set, and in this example this variable gets set to 0 because that's the default in the configure script if cross compiling is detected ).
> 
> I've "solved" this by adding a configure section to the ntpd Makefile, thus:
> 
> ```
> define Build/Configure
>     $(call Build/Configure/Default, $(CONFIGURE_VARS))
> endef
> ```
> 
> Presumably this works because the variables are embedded into config.status and therefor handled as arguments. However, this doesn't explain WHY this happens, and why it might be that most of the time it seems to work ( well, I've never got it to work on Ubuntu 24.10, but really haven't had many attempts and not tried compilation with a clean .config because time is limited, but I've never had an issue in the past and have compiled OpenWrt MANY times ).
> 
> Any ideas as to the root cause? In the meantime hopefully this workaround may help if someone else has the issue.
> 
> Could someone re-open the issue please? I don't really want to create a new one since it's identical and this was created recently, implying that the cause is fairly recent even though there's obviously a trigger for it somewhere.

Followed and it works. It just isn't the cache problem.

tofurky, 2025-02-06:
Both adding `Build/Configure` and applying this (to my local package repository which contains ntpd) fixed it: https://github.com/openwrt/packages/commit/de0e7efdb8e8dc925184fa8495819be0dcbb3475

Went with the above commit as that's what's in master and 24.10. It hasn't been backported to the 23.05 branch, though.