openwrt/packages #25402: python-certifi: Compilation fails

tiagogaspar8, 2024-11-18:
Maintainer: Eneas U de Queiroz <cotequeiroz@gmail.com> (@cotequeiroz)
Environment: Archer c7, built on x86 Fedora 41

Description:

When building I face the following error:
```
rm -f /mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/.built
touch /mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/.built_check
cd "/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2" && CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -DPIC -fpic" CXX="mips-openwrt-linux-musl-g++" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2=certifi-2024.2.2 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/usr/include -I/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include -I/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include/fortify -I/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl/usr/include/python3.11" LDFLAGS="-L/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/usr/lib -L/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lpython3.11" _PYTHON_HOST_PLATFORM="linux-mips" _PYTHON_SYSCONFIGDATA_NAME="_sysconfigdata__linux_mips-linux-muslsf" PYTHONPATH="/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl/usr/lib/python3.11:/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl//usr/lib/python3.11/site-packages:/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/ipkg-install//usr/lib/python3.11/site-packages" PYTHONDONTWRITEBYTECODE=1 _python_sysroot="/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl" _python_prefix="/usr" _python_exec_prefix="/usr"  CARGO_BUILD_TARGET=mips-unknown-linux-musl CARGO_HOME=/mnt/openwrt/ss_mod/dl/cargo CARGO_PROFILE_RELEASE_CODEGEN_UNITS=1 CARGO_PROFILE_RELEASE_DEBUG=false CARGO_PROFILE_RELEASE_DEBUG_ASSERTIONS=false CARGO_PROFILE_RELEASE_LTO=true CARGO_PROFILE_RELEASE_OPT_LEVEL=z CARGO_PROFILE_RELEASE_OVERFLOW_CHECKS=true CARGO_PROFILE_RELEASE_PANIC=unwind CARGO_PROFILE_RELEASE_RPATH=false CARGO_TARGET_MIPS_UNKNOWN_LINUX_MUSL_LINKER=mips-openwrt-linux-musl-gcc RUSTFLAGS="-Ctarget-feature=-crt-static -lssp_nonshared -Clink-arg=-fuse-ld=bfd" TARGET_CC=mips-openwrt-linux-musl-gcc TARGET_CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2=certifi-2024.2.2 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " PYO3_CROSS_LIB_DIR="/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl/usr/lib/python3.11" SETUPTOOLS_RUST_CARGO_PROFILE="release"    /mnt/openwrt/ss_mod/staging_dir/hostpkg/bin/python3.11  -m build --no-isolation --outdir "/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/"/openwrt-build --wheel   "/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/" 
* Getting build dependencies for wheel...
/mnt/openwrt/ss_mod/staging_dir/hostpkg/lib/python3.11/site-packages/setuptools/dist.py:530: UserWarning: Normalizing '2024.02.02' to '2024.2.2'
  warnings.warn(tmpl.format(**locals()))
running egg_info
writing certifi.egg-info/PKG-INFO
writing dependency_links to certifi.egg-info/dependency_links.txt
writing top-level names to certifi.egg-info/top_level.txt
reading manifest file 'certifi.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files found matching '.github/'
warning: manifest_maker: MANIFEST.in, line 4: 'recursive-exclude' expects <dir> <pattern1> <pattern2> ...

adding license file 'LICENSE'
writing manifest file 'certifi.egg-info/SOURCES.txt'
* Building wheel...
/mnt/openwrt/ss_mod/staging_dir/hostpkg/lib/python3.11/site-packages/setuptools/dist.py:530: UserWarning: Normalizing '2024.02.02' to '2024.2.2'
  warnings.warn(tmpl.format(**locals()))
running bdist_wheel
running build
running build_py
running egg_info
writing certifi.egg-info/PKG-INFO
writing dependency_links to certifi.egg-info/dependency_links.txt
writing top-level names to certifi.egg-info/top_level.txt
reading manifest file 'certifi.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files found matching '.github/'
warning: manifest_maker: MANIFEST.in, line 4: 'recursive-exclude' expects <dir> <pattern1> <pattern2> ...

adding license file 'LICENSE'
writing manifest file 'certifi.egg-info/SOURCES.txt'
warning: build_py: byte-compiling is disabled, skipping.

installing to build/bdist.linux-mips/wheel
running install
running install_lib
creating build/bdist.linux-mips/wheel
creating build/bdist.linux-mips/wheel/certifi
copying build/lib/certifi/__init__.py -> build/bdist.linux-mips/wheel/certifi
copying build/lib/certifi/__main__.py -> build/bdist.linux-mips/wheel/certifi
copying build/lib/certifi/core.py -> build/bdist.linux-mips/wheel/certifi
copying build/lib/certifi/cacert.pem -> build/bdist.linux-mips/wheel/certifi
copying build/lib/certifi/py.typed -> build/bdist.linux-mips/wheel/certifi
warning: install_lib: byte-compiling is disabled, skipping.

running install_egg_info
Copying certifi.egg-info to build/bdist.linux-mips/wheel/certifi-2024.2.2-py3.11.egg-info
running install_scripts
creating build/bdist.linux-mips/wheel/certifi-2024.2.2.dist-info/WHEEL
creating '/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/openwrt-build/tmp0nyt3z_9/certifi-2024.2.2-py3-none-any.whl' and adding 'build/bdist.linux-mips/wheel' to it
adding 'certifi/__init__.py'
adding 'certifi/__main__.py'
adding 'certifi/cacert.pem'
adding 'certifi/core.py'
adding 'certifi/py.typed'
adding 'certifi-2024.2.2.dist-info/LICENSE'
adding 'certifi-2024.2.2.dist-info/METADATA'
adding 'certifi-2024.2.2.dist-info/WHEEL'
adding 'certifi-2024.2.2.dist-info/top_level.txt'
adding 'certifi-2024.2.2.dist-info/RECORD'
removing build/bdist.linux-mips/wheel
Successfully built certifi-2024.2.2-py3-none-any.whl
cd "/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2" && CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -DPIC -fpic" CXX="mips-openwrt-linux-musl-g++" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2=certifi-2024.2.2 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/usr/include -I/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include -I/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/include/fortify -I/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl/usr/include/python3.11" LDFLAGS="-L/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/usr/lib -L/mnt/openwrt/ss_mod/staging_dir/toolchain-mips_24kc_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lpython3.11" _PYTHON_HOST_PLATFORM="linux-mips" _PYTHON_SYSCONFIGDATA_NAME="_sysconfigdata__linux_mips-linux-muslsf" PYTHONPATH="/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl/usr/lib/python3.11:/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl//usr/lib/python3.11/site-packages:/mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/ipkg-install//usr/lib/python3.11/site-packages" PYTHONDONTWRITEBYTECODE=1 _python_sysroot="/mnt/openwrt/ss_mod/staging_dir/target-mips_24kc_musl" _python_prefix="/usr" _python_exec_prefix="/usr"  CARGO_BUILD_TARGET=mips-unknown-linux-musl CARGO_HOME=/mnt/openwrt/ss_mod/dl/cargo CARGO_PROFILE_RELEASE_CODEGEN_UNITS=1 CARGO_PROFILE_RELEASE_DEBUG=false CARGO_PROFILE_RELEASE_DEBUG_ASSERTIONS=false CARGO_PROFILE_RELEASE_LTO=true CARGO_PROFILE_RELEASE_OPT_LEVEL=z CARGO_PROFILE_RELEASE_OVERFLOW_CHECKS=true CARGO_PROFILE_RELEASE_PANIC=unwind [... cut ...]
    return self.write_to_fs(scheme, path_, stream, is_executable)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/openwrt/ss_mod/staging_dir/hostpkg/lib/python3.11/site-packages/installer/destinations.py", line 167, in write_to_fs
    raise FileExistsError(message)
FileExistsError: File already exists: /mnt/openwrt/ss_mod/build_dir/target-mips_24kc_musl/pypi/certifi-2024.2.2/ipkg-install/usr/lib/python3.11/site-packages/certifi/__init__.py
```
I tried to troubleshoot but not sure what to look for.

gretel, 2024-12-16:
same issue here. in some cases cleansing first is not energy efficient.

commodo, 2026-04-05:
I think this was fixed recently with
https://github.com/openwrt/packages/pull/27787/changes/e8c908bcd3c03794b571ff8f813b101129154bd9

It's in 25.12 and master.

Please test and re-open if still an issue